verilator/verilator #3146: Incorrect UNOPTFLAT warning

rachitnigam, 2021-09-22:
The following program generates an UNOPTFLAT warning which I think is incorrect. 

```systemverilog
module main (
    input logic go
);
    logic [31:0] add10_left;
    logic [31:0] add10_right;
    logic [31:0] add10_out;
    logic [31:0] add9_left;
    logic [31:0] add9_right;
    logic [31:0] add9_out;

    assign add10_out = add10_left + add10_right;
    assign add9_out = add9_left + add9_right;

    assign add10_right =
      go ? add9_out :
     ~go ? 32'd1 : 32'd0;

    assign add9_right =
      go ? 32'd1 :
     ~go ? add10_out : 32'd0;
endmodule
```

When `go = 1`, we'll have the assignments:
```
add10_right = add9_out;
add9_right = 32'd1;
```
and when `go = 0`, we'll have:
```
add10_right = 32'd1;
add9_right = add10_out;
```

Both of which don't seem to create circular logic (correct me if I'm wrong about this). 

Version: Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)

Would you be willing to try to fix Verilator yourself with assistance?
Yes, depending on the scope of the problem

wsnyder, 2021-09-22:
If you follow the statements in the order that are printed in the error it does look circular to me. Do you disagree with one of these steps?:

                ex.v:12:25:      Example path: add9_right
                ex.v:16:21:      Example path: ASSIGNW
                ex.v:13:25:      Example path: add9_out
                ex.v:18:24:      Example path: ASSIGNW
                ex.v:9:25:       Example path: add10_right
                ex.v:15:22:      Example path: ASSIGNW
                ex.v:10:25:      Example path: add10_out
                ex.v:22:23:      Example path: ASSIGNW
                ex.v:12:25:      Example path: add9_right

It doesn't look at go being true or false, it must be non-circular for both.  If go is a parameter, then that would be different and it could be non-circular.

rachitnigam, 2021-09-22:
Maybe there is a gap in my understanding Verilog's semantics then. Is there any way to write the program with the intent that there is `add10 -> add9` chain computation when `go = 1` and a `add9 -> add10` chain when `go = 0` without running into this problem? Would using `always_comb ... if (go) ... else if (~go)` fix this?

wsnyder, 2021-09-23:
An if statement wouldn't be any better.  I don't understand what you're trying to do, write the logic as if it was a schematic, from e.g. start to end, without using "backtracking" through certain pieces of logic you've already gone through again.

rachitnigam, 2021-09-23:
Got it. Unfortunately the code is generated by a compiler upstream. This is just a minimal test case I got by reducing the offending program. Anyways, thanks for the help!

closed 2021-09-23 by rachitnigam