verilator/verilator #3147: Generating waveforms for in C++

amfarla, 2021-09-24:
I am using Verilator 4.110 2021-02-25 rev v4.110-15-g702140bf.

My goal is to produce a *.vcd file to generate wave forms in C++, as described here: https://veripool.org/guide/latest/faq.html. I am following the steps shown in this FAQ section, but I am getting the error depicted when compiling my simulated cpu core. I cannot seem to find definitions for the `trace` method anywhere within the verilator source code. The `main` function which is attempting to use `trace` is also shown below.

[image]

[image]

I have tried replacing the `trace` method call with `Verilated::debug(9)` to generate a *.vcd file. When I do this, my core compiles and I can run the executable produced with the `--debug` command. While the core executes as expected, the *.vcd file remains empty. 

In summary, I would like to be able to generate waveforms for my C++ but I cannot seem to do so successfully from the documentation I have found thus far. 

Thank you for your help!

wsnyder, 2021-09-24:
Well, the easiest way is just to do it in verilog and then you don't need any C++ code:

   initial begin
      $dumpfile(filename);
      $dumpvars();
   end

But as to your example, it looks like you didn't pass "--trace" when you called verilator to make your model.

closed 2021-09-24 by wsnyder

amfarla, 2021-09-24:
Thank you for your quick response and help.