verilator/verilator #3148: An internal error occurred when using clang 12 or 13

kuopinghsu, 2021-09-25:
Here is the sample Verilog code.

```
module test(
    input           clk,
    input           resetb,
    input           stall,
    input    [ 4:0] sel,
    input    [31:0] datain
    );

reg [31: 0] regs [31: 0];
integer i;

always @(posedge clk or negedge resetb) begin
    if (!resetb) begin
        for(i = 0; i <= 31; i = i + 1) regs[i] <= 32'h0;
    end else if (!stall) begin
        regs[sel] <= datain;
    end
end

endmodule
```

If I use the following command to build Verilator with clang 12 or 13

CC=clang CXX=clang++ ./configure && make install

I got the following internal error

$ verilator -cc test.v
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt

If I build Verilator with clang 11, there is no issue.

wsnyder, 2021-09-25:
Well clang made this one annoying, there was a call to "nodep->debugTreeChange" and although debugTreeChange was an empty function because it's illegal to call members on a null nodep, clang optimized away a later nullptr compare.

Looking for other similar issues...  will close if I don't find any.

wsnyder, 2021-09-25:
All tests pass.

closed 2021-09-25 by wsnyder