verilator/verilator #3151: How to get the Toggle Coverage and Functional Coverage

yuqi-ali, 2021-09-27:
I add -CFLAGS -DCOVERAGE --coverage while compile rtl, and get the coverage.dat after simulation.
After generate html report, there are no toggle coverage and functional coverage ,how to display it

wsnyder, 2021-09-27:
Sounds like the right process. Is there information in the coverage.dat file?  When you say "no coverage" do you mean no coverage points, or the points are zero?

yuqi-ali, 2021-09-28:
I can get the line coverage，but no toggle coverage and functional coverage.
after get the .dat file. I use these commands to generate html report
verilator_coverage -write-info coverage.info coverage.dat
genhtml coverage.info --output-directory coverage

wsnyder, 2021-09-28:
Sorry, but you need to debug it, you need to please give a lot more information to determine what is wrong.

Can you submit a standalone example?  What sort of user coverage statements are there in the code? And are there "toggle" statements in the .dat file?  Does the verilator_coverage text output show the coverage information correctly (ignoring genhtml)?

yuqi-ali, 2021-09-28:
Do you have some example about this? I use Verilator in SpinalHDL. And SpinalHDL

yuqi-ali, 2021-09-28:
SpinalHDL can Simulator by using Verilator automatically

wsnyder, 2021-09-28:
There's an example here: https://github.com/verilator/example-systemverilog

yuqi-ali, 2021-09-28:
I test this example in Verilator 4.200 ,it seems that the Toggle Coverage and Functional Coverage still not get:
Overall coverage rate:
  lines......: 80.0% (16 of 20 lines)
  functions..: no data found

wsnyder, 2021-09-28:
The coverage is intentionally not 100% to demonstrate what misses look like.

yuqi-ali, 2021-09-28:
There are no messages about toggle coverage, is it right?

wsnyder, 2021-09-28:
The coverage on lines with signal declarations are toggle coverage.

wsnyder, 2021-09-28:
(I assume you are referring to the example I referenced)

wsnyder, 2021-10-01:
Closing as think answered, but feel free to follow up.

closed 2021-10-01 by wsnyder

stu-e, 2023-03-08:
I have only just started looking into coverage and @yuqi-ali is correct in that 'toggle coverage' is missing. Simulators like Questa will generate reports showing which logic bits have toggled state and which have not changed state. So if you have a vector of 4 bits say and the vector only ever takes on the values of 0 and 1 then bit 0 has toggled but bits 3-1 have not. This shows part of the design has not been exercised.

wsnyder, 2023-03-08:
Verilator collects toggle coverage, and with its own reports shows it, so I don't believe there is any issue here.  @stu-e if you see issues with verlator-generated toggle reports please open a new issue.

wsnyder, 2023-03-08:
[Using the most recent version that is]

stu-e, 2023-03-08:
Forgive me I'm stumped. I see what looks like toggle coverage statistics inside file coverage.dat but beyond that the post processing outlined at https://github.com/verilator/example-systemverilog/blob/master/Makefile#L66 does not produce anything showing toggle coverage. Please what am I missing?

wsnyder, 2023-03-08:
You can't use the C-tool coverage as it doesn't know what toggle is.  Use Verilator's tools, see the Verilator documentation and Verilator's test_regress/t/t_cover_toggle_points.out

Faisalawan11, 2023-06-01:
> I can get the line coverage，but no toggle coverage and functional coverage. after get the .dat file. I use these commands to generate html report verilator_coverage -write-info coverage.info coverage.dat genhtml coverage.info --output-directory coverage

Hello yuqi-ali, which command did you use for getting .dat file?

Waiting your response 

Thank you