verilator/verilator #3152: Is it possible to trace signals outside the verilated module?

it3q, 2021-09-29:
I use verilator to make "unit test" like things on Verilog modules. In this setup there is a Verilog module to test (DUT) and a stimulus that generates input for DUT and checks the response for that. One part of the stimulus is written in C++ (using SystemC) to be reusable, easier to maintain, etc. This code (SystemC module / C++ code) is compiled and linked with the DUT into one executable which runs fine, signals in verilated module can be traced as well. 

However, there are cases when it would be nice to see some internal signals from the stimulus part in the traced vcd file (usually to fix issues in the input generator site).

Is it somehow possible to add SystemC signals outside from the DUT to the trace? Or even generally, is it possible to record a C++ variable in the vcd waveform along with the DUT signals?

If the answer is no, how would you debug the stimulus code in such a situation?

(I could move the functionality of stimulus to Verilog modules of course. Then verilate that one as well along with the DUT, run and trace. It would surely work, but if possible I would avoid that.)

wsnyder, 2021-09-29:
SystemC itself (without Verilator) has VCD trace capabilities, see the SystemC documentation.  Use those, and have Verilator make it's VCD, then in your waveform viewer just open both files - many viewers then let you mix and match signals from multiple traces on one display.

closed 2021-09-29 by wsnyder

it3q, 2021-09-30:
Oh, that's nice! :) Thank you for the quick answer!