verilator/verilator #3153: `hier` data is empty in calls to VL_FINISH_MT or VL_STOP_MT

ekiwi, 2021-09-29:
When there is an assertion or a finish or fatal call in my Verilog, code, these get translated into call to either `VL_STOP_MT` or `VL_FINISH_MT. For example:

```.cpp
    if (VL_UNLIKELY(((IData)(vlTOPp->selParent) & (IData)(vlTOPp->stop0En)))) {
        VL_FINISH_MT("test.sv", 93, "");
    }
    if (VL_UNLIKELY(((IData)(vlTOPp->selParent) & (IData)(vlTOPp->stop1En)))) {
        VL_WRITEF("[%0t] %%Error: test.sv:104: Assertion failed in %Ntest\n",
                  64,VL_TIME_UNITED_Q(1),vlSymsp->name());
        VL_STOP_MT("test.sv", 104, "");
    }
```

The first two argument (`filename` and `linenum`) contain the expected values. However the third argument `hier` is always left blank. I was hoping that it would contain a hierarchical path in order to be able to know which instance of a module had the failing assertion. Was that the intention and it was just never implemented? Or am I misunderstanding the use of `hier`?

What 'verilator --version' are you using?  Did you try it with the git master version?

I tried it out with the version that comes with Fedora (`4.108`). I also tested the latest version from the [oss-cad-toolchain](https://github.com/YosysHQ/oss-cad-suite-build/releases) (`Verilator 4.213 devel rev v4.212-46-g20af8a36`) which also leaves `hier` blank.

Would you be willing to try to fix Verilator yourself with assistance?

If I understand correctly that `hier` should actually contain the hierarchical path, similar to what is already included in the error message (using `VL_WRITEF`), I could have a shot at fixing that.

wsnyder, 2021-10-01:
Yes, hier is a holdover from some SystemC stuff, and not currently passed in.  I see the usefulness for passing the scope on stop/finish, but the caution would be that if the code uses this in a low-level module that gets inlined it will result in the code not being identical (because each instance will have a different scope name).  The result will be more code and longer compiles.  I suspect though in many cases there's a earlier print that already has the scope, so this expansion won't matter for much.

To add support for this you'll need to create an AstScopeName below each Stop/Finish, as that is how the proper name gets tracked through inlining.  Then in V3Emit print it out.

wsnyder, 2021-12-12:
I think it's unlikely this will be a priority to work on so closing it out.  If you'd or someone wants to consider a pull request that would be fine.

closed 2021-12-12 by wsnyder