verilator/verilator #3154: Dynamic cast in class method provokes error "$time has no units"

mballance, 2021-10-01:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
The testcase is derived from 't_castdyn.v' in the current test suite:

```
package pkg;
  class Base;
  endclass
  class BasedA extends Base;
  endclass
  class BasedB extends Base;
  endclass

  function automatic doit(Base b);
          BasedA bb;
          $cast(bb, b); // Okay
  endfunction

  class foo;
        function doit(Base b);
                BasedA bb;
                $cast(bb, b); // Not okay
        endfunction
  endclass

endpackage

module t (/*AUTOARG*/);
   import pkg::*;
   int i;
   int a;
   int ao;

   Base b;
   Base bo;
   BasedA ba;
   BasedA bao;
   BasedB bb;
   BasedB bbo;

   // ...
endmodule
```

The dynamic casts of classes in the 'initial' block in the module work fine, as does the dynamic cast in the package-level function.  The dynamic cast in the class method produces the following error:

```
%Error: Internal Error: t/t_castdyn.v:24:3: ../V3EmitCFunc.h:862: $time has no units
                                          : ... In instance pkg::foo
   24 |   $cast(bb, b);
      |   ^~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

What 'verilator --version' are you using?  Did you try it with the git master version?
I'm using 4.212

Would you be willing to try to fix Verilator yourself with assistance?
Absolutely. Give me a hint and I'm happy to submit a PR!

wsnyder, 2021-10-01:
Thanks for the report, I just confirmed your example is already fixed in master, see #3116.

closed 2021-10-01 by wsnyder

mballance, 2021-10-01:
Thanks for the fast response, @wsnyder, and apologies for the spam. Glad the issue is fixed, and happy to move to 'main'.

wsnyder, 2021-10-01:
No worries - a well written bug report is never spam ;)