verilator/verilator #3155: [verilator_profcfunc] UnicodeDecodeError: 'utf-8' codec can't decode

drom, 2021-10-01:
```
verilator --version
Verilator 4.213 devel rev v4.212-46-g20af8a36
```

I have verilated / simulated my Verilog code with `--prof-cfuncs` option.

I got `gmon.out` file. That looks like binary file.

Running: `verilator_profcfunc gmon.out` produces the following error:

``` 
Traceback (most recent call last):
  File "/scratch/drom/verilator/bin/verilator_profcfunc", line 199, in <module>
    profcfunc(Args.filename)
  File "/scratch/drom/verilator/bin/verilator_profcfunc", line 18, in profcfunc
    for line in fh:
  File "/usr/lib64/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb0 in position 21: invalid start byte
```

wsnyder, 2021-10-01:
verilator_profcfunc takes in the output from "gprof" not the raw gmon.out. Please see the examples.  If the documentation is missing this step or needs other clarifications please consider filing a pull request to fix it or add other suggestions, thanks.

closed 2021-10-01 by wsnyder

drom, 2021-10-02:
The documentation is fine. It was my surface reading skills failure.