verilator/verilator #3157: Spurious warning about "circular logic" in a full adder chain

mcandre, 2021-10-05:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

This 3-bit full adder implementation.

Source:

https://github.com/mcandre/toys/blob/master/rtl/systemverilog/wholeass/test_wholeass.sv#L1

Which passes this test suite:

https://github.com/mcandre/toys/blob/master/rtl/systemverilog/wholeass/main.cpp

When the line `/* verilator lint_off UNOPTFLAT */` is cleared out from the design, then Verilator reports this trace:

```console
%Warning-UNOPTFLAT: wholeass.sv:1:46: Signal unoptimizable: Feedback to clock or circular logic: 'test_wholeass.__Vcellout__genblk1[1].wa____pinNumber4'
                                    : ... In instance test_wholeass
    1 | module wholeass(input carry_in, a, b, output carry_out, sum);
      |                                              ^~~~~~~~~
                    test_wholeass.sv:4:1: ... note: In file included from test_wholeass.sv
                    ... For warning description see https://verilator.org/warn/UNOPTFLAT?v=4.213
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
                    wholeass.sv:1:46:      Example path: test_wholeass.__Vcellout__genblk1[1].wa____pinNumber4
                    test_wholeass.sv:19:58:      Example path: ASSIGNW
                    test_wholeass.sv:12:18:      Example path: test_wholeass.carries
                    wholeass.sv:3:22:      Example path: ASSIGNW
                    wholeass.sv:1:46:      Example path: test_wholeass.__Vcellout__genblk1[1].wa____pinNumber4
%Error: Exiting due to 1 warning(s)
```

Note that the generate syntax is not the problem: The same spurious error message appears when the generator expression is replaced with a manual, boilerplate hookup between the internal carry bits.

What 'verilator --version' are you using?  Did you try it with the git master version?

Current git master.

```console
$ verilator --version
Verilator 4.213 devel rev v4.212-48-g2560fc86
```

Would you be willing to try to fix Verilator yourself with assistance?

I'm very new to RTL programming :P

Here's a thought, maybe the circular logic scanner just doesn't take into account array indices. It's possible that the linter is confusing multiple reads and/or writes to the one array variable. Actually, we have three: The distinct array elements.

wsnyder, 2021-10-05:
>Maybe the circular logic scanner just doesn't take into account array indices.

That's correct.  If you look at the documentation that it suggests the inidices limitation is described there: https://verilator.org/warn/UNOPTFLAT

Note it also describes how to use split_var to get around this.  However generally the code can be better written to avoid such dependencies.

closed 2021-10-05 by wsnyder

mcandre, 2021-10-05:
Interestingly, the documentation for this linter warning recommends enabling another hint annotation, `/*verilator split_var*/`.

Anything wrong with verilator automatically applying split_var in all cases where it detects this performance concern?

wsnyder, 2021-10-05:
It doesn't automatically do split_var because it is better to fix the code, and more selfishly by the time this is detected it is too hard to then make the change.

mcandre, 2021-10-05:
Alright, understood.

I want to make sure that I understand the documentation better. Could you please help me to manually split my variable?

I tried some syntax based on the example code at https://verilator.org/guide/latest/warnings.html#cmdoption-arg-UNOPTFLAT , but so far I am unable to come up with the right thing to satisfy the design parser, oy!

wsnyder, 2021-10-05:
Please indicate what you tried that didn't work.  My guess is you put the comment after instead of before the semicolon.

mcandre, 2021-10-05:
Oh, the comment works fine. I'm asking for the right plain SystemVerilog syntax to manually split the variables.

The documentation offers a terse tip about creating a second variable xout, as in `wire [2:0] xout = {x[1:0], shift_in};`.

I tried that for my full adder, substituting N for 2. And my N is set to 3. And following the doc, I configured my receiving logic to use the manually split variable.

```systemverilog
`include "wholeass.sv"

module test_wholeass;
    parameter N = 3;
    reg [N-1:0] a /* verilator public */,
                b /* verilator public */;
    reg carry_in /* verilator public */;
    wire carry_out /* verilator public */;
    wire [N-1:0] sum /* verilator public */;

    wire [N-2:0] carries;
    wire [N-2:0] carry_outs = { carries[N-2:0] };
    wholeass wa_head(carry_in, a[0], b[0], carries[0], sum[0]);
    wholeass wa_tail(carry_outs[N-2], a[N-1], b[N-1], carry_out, sum[N-1]);
    genvar i;

    generate
        for (i = 1; i < N - 1; i++)
            wholeass wa(carry_outs[i-1], a[i], b[i], carries[i], sum[i]);
    endgenerate
endmodule
```

However, verilator reports this syntax is circular logic. So I'm kind of lost here for the right way to do this.

My GitHub main branch is using the comment automatic split var syntax for the time being. But I would like to learn more about how to do this manually, so that my code will be portable to other simulators.

wsnyder, 2021-10-05:
I think any solution with a generate will probably result in the warning, so you would need to use split_var.  However if you can remove the whoass module and code it as a normal assignment with the parameter setting the width, rather than a generate, that would likely solve the issue.

I don't know what logic is in the submodule but at first glance it looks like a carry propagation, which seems coded at an unnecessary low level. In most cases using much higher abstraction e.g. "assign {carry,sum{N-1:0]} = a + b + carry_in;" will typically get better results in synthesis and simulation.