verilator/verilator #3166: 'No dtype on node with hasDType()' internal error

mballance, 2021-10-11:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
When running without --debug, this results in a crash. When running in the regression suite, I receive the error above.
This appears to result from at least two scenarios:
- Passing a scoped enumerator to a class constructor
- Calling the constructor for the current class and passing local fields to the constructor (pattern: clone method)

I've updated two current regression-suite tests to show the error.

## t_class_enum.v

```
module t (/*AUTOARG*/);

class Cls;
   typedef enum {A = 10, B = 20, C = 30} en_t;
   en_t m_en;

   function new(en_t v);
       m_en = v;
   endfunction
endclass

   initial begin
      Cls c;
      if (c.A != 10) $stop;
      c = new(Cls::B);
      if (c.m_en != 20) $stop;
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

Addition of the `new' method and call to constructor triggers the error.

## t_class_new.v
Modifying Cls2Arg by adding the `clone` method triggers the error.
```
class Cls2Arg;
   int imembera;
   int imemberb;
   function new(int i, int j);
      imembera = i + 1;
      imemberb = j + 2;
   endfunction

   function Cls2Arg clone();
      Cls2Arg ret;
      ret = new(imembera, imemberb);
      return ret;
   endfunction
endclass
```

What 'verilator --version' are you using?  Did you try it with the git master version?
I'm using 'master' as of 20211010 updated 20:05 PDT.

Would you be willing to try to fix Verilator yourself with assistance?
Absolutely. I think it would be helpful to understand where a node's DType is expected to be set, and possible reasons for it not being set.

wsnyder, 2021-10-13:
Thanks for the great test - simple fix.

closed 2021-10-13 by wsnyder

mballance, 2021-10-13:
Many thanks, @wsnyder!