verilator/verilator #3168: FST dumping is still very slow and creates waveform files that are much larger than expected

tomverbeure, 2021-10-11:
# Problem description

Project: https://github.com/tomverbeure/cxxrtl_eval

Verilator dumping VCD:

* 4s simulation time
* 2GB VCD file

Verilator dumping FST:

* 100s simulation time
* 1GB FST file

Converting the VCD file to FST with `vcd2fst`:
* 12s
* 50MB FST file

For some reason, FST dumping from Verilator is very slow and creates files that are much larger than they need to be.

I'm aware of issue 1566, but a factor 20 in simulation speed is still way too much.

# Version

Verilator version: 4.212

# Repro instructions

```
git clone  https://github.com/tomverbeure/cxxrtl_eval
cd cxxrt_eval/verilator
make
time ./tb_vcd > /dev/null
time ./tb_fst > /dev/null
```

wsnyder, 2021-10-11:
Perhaps you would be willing to profile the code and see where the time is going?

tomverbeure, 2021-10-11:
I'll give it a try.

tomverbeure, 2021-10-11:
On a hunch, I changed the following assignment to 0:

https://github.com/verilator/verilator/blob/9731ec76e3d1c77cdb782bf9e55936723795c3f2/include/gtkwave/fstapi.c#L1916

After that, simulation time went from 100s to 5s, and file size went down from 1GB to the expected 20MB. (I reported 50MB in my initial comment for the vcd2fst converted FST file. That was wrong. It should have been 20MB.)

I don't understand the reasons to decide to flush or not flush, but this should be enough of a lead for somebody who knows the internals quickly figure out what's going one.

Tom

wsnyder, 2021-10-13:
@gtkwave any comment on the above?

gtkwave, 2021-10-13:
That's not a lot to go on.  Very slow and much larger compared to what?
-Tony

wsnyder, 2021-10-14:
Tony/@gtkwave, A VCD then vcd2fst gives a smaller file, and there was a suggested patch - please see comments above this one, thanks.  Specifically is the patch right, and if so can you put it in your upstream?  If not can you suggest other things to try/fix?