verilator/verilator #3173: Giving the SystemC module the same name as the top level verilog module using --preifx results in not compilable code

voertler, 2021-10-14:
Verilator supports setting the name of the generated SystemC module using the --prefix option. This leads to compile errors when the prefix is the same as the verilog module, because a member pointer of the same name as the class is generated.

```
module myflop (input clk, r, d, output logic q);
  // dummy member for access
  logic [7:0] mem[7] /* verilator public */;
  always_ff @(posedge clk or posedge r)
    if (r)
      q <= 0;
    else
      q <= d;
endmodule
```

The generated code looks something like this when verilating using `verilator --sc --prefix myflop --pins-sc-uint myflop.v` 

```
...
SC_MODULE(myflop) {
  public:
    // CELLS
    // Public to allow access to /*verilator_public*/ items;
    // otherwise the application code can consider these internals.
    myflop_myflop* myflop;
...
```
This is illegal and will lead to C++ compile errors. Is there a way to rename the pointer `myflop_myflop* myflop`? Starting from Verilator 4.210 it seems that the pointer is always generated even when not using /*verilator_public*/.

Would you be willing to try to fix Verilator yourself with assistance?
Yes

wsnyder, 2021-10-14:
Yes, we seemed to have lost the PVT insertion that was previously applied in V3Name when no publics are present.

You indicated you'd be willing to look at a fix, that would be appreciated.  Things to put into a pull request to fix this:

- [ ] Make a test, see if fail until fix below
- [ ] See if V3Name is properly renaming the object by looking at .debug files - it probably is, if so see why it isn't properly propagating to the emit code (V3Emit).  If not see why not.

gezalore, 2021-10-17:
I would have expected it was #3036 that broke this. I guess it's not renamed due to this:
https://github.com/verilator/verilator/blob/ac622f9db8675cc2271aa1b08f2a3a7232c10b28/src/V3Name.cpp#L97-L102
As you have a `/* verialtor public */` variable (the `mem` array), the top level module will not be inlined nor renamed because you are supposed to be able to access it as `<model_pointer>->myflop->mem`. 

I don't see a way of fixing this in a backwards compatible way as either the model name, or the instance name needs to change.

Note using `/* verilator public_flat */` would likely make this work but you will have to change the C++.

wsnyder, 2024-11-12:
As last comment pointed out given the use case it's likely changing the main C++ code is the only likely option.  At this point I'm assuming we don't have to update anything; @voertler reopen if you still see this as an issue, thanks.