verilator/verilator #3175: Usage of attributes clocker and clock_enable

gezalore, 2021-10-17:
I am really struggling with the clocker and clock_enable attributes.

Let's start with clock_enable: Reading V3Order all this seems to achieve is ignore a the dependency in combinational logic on the annotated signal. Under what circumstance is this useful (and does not kill you by ignoring the dependency)? I cannot think of a use case.

As of clocker: This seems to do two things:
1. Make some dependencies [as defined by whatever `OrderClkMarkVisitor` propagates this attribute to, which is by no means comprehensive] feeding this variable through combinational logic be a hard edge, instead of the a cutable one. The point of this is that we don't have a change detect on the generated clock, which seems fair, but Verilator could compute this itself without any annotation.
2. Similar to clock_enable, ignore all *other* variable dependencies in combinational logic driving the clocker.

For the "let's not have the generated clock circular" part (which seems like just a performance thing?), shouldn't we just change all edges that feed a sensitivity list (i.e.: clocks) to non-cutable in the V3Order dependency graph. If we end up with a hard cycle, throw up hands, or warn and automatically pick an edge to cut in some intelligent way?

As of the ignored dependencies, when are these useful? They seem always unsafe.

wsnyder, 2021-10-17:
Fundamentally this stems from Verilator following synthesis rules and not having timing nor following the scheduling semantics.

If you have flops with a clock generated in between:

       input clk;
       wire clk2 = clk & gater2;
       wire #1 clk3 = clk & gater3;
       always_ff @(posedge_clk) d <= ...
       always_ff @(posedge_clk2) d2 <= d
       always_ff @(posedge_clk3) d3 <= d

In delay-honoring simulators, clk and clk2 are evaluated in the same timestep and d will not race to d2.  In contrast clk3 follows clk, and d will race to d2.

Verilator ignores the pound delays, and assumes all logic has delay (like synthesis).  clk2 is considered generated, as is clk3. Signals will race from d to d2, and d to d3, as they would in synthesis unless you put a timing constrant (that cannot be expressed in verilog) saying that the clk2 network has same timing as clk3.  Interestingly in synthesis you can also say that clk3 has the same timing as clk2 - that is timing is also ignored and the synthesis constraints are what matter.

clocker and clk_enable were hacks to express the synthesis constraints, and/or get around lack of delays.

So in summary, perhaps the default of what they do is wrong, and perhaps there are better ways to express intent or fix this, but they certainly are necessary.  I'd argue any generated clock is dangerous, and the right thing is probably to require every generated clock have an attribute to make clear what is expected timing-wise.

gezalore, 2021-10-17:
> Verilator ignores the pound delays, and assumes all logic has delay (like synthesis). clk2 is considered generated, as is clk3. Signals will race from d to d2,

Will they?

I just fed through this the tip of master:
```
module top(
   input wire clk,
   input wire en,
   input wire i,
   output reg d,
   output reg d2
);
  wire clk2 = clk & en;

  always_ff @(posedge clk)  d  <= i;
  always_ff @(posedge clk2) d2 <= d;
endmodule
```
I got this out (simplified):
```
void Vtop___024root___eval(Vtop___024root* vlSelf) {
    vlSelf->top__DOT__clk2 = vlSelf->clk & vlSelf->en;
    if (vlSelf->top__DOT__clk2 & ~vlSelf->__Vclklast__TOP__top__DOT__clk2) {
        vlSelf->d2 = vlSelf->d;
    }
    if (vlSelf->clk & ~vlSelf->__Vclklast__TOP__clk) {
        vlSelf->d = vlSelf->i;
    }

    vlSelf->__Vclklast__TOP__clk = vlSelf->clk;
    vlSelf->__Vclklast__TOP__top__DOT__clk2 = vlSelf->top__DOT__clk2;
}
```
Semantically that's the two clocks happening together, i.e.: d does not race to d2. (`d2` will be old `d`, `d` will be old `i`). Why does this does not match what you were saying above?

gezalore, 2021-10-17:
BTW I'm just trying to formulate a mental model about what the actual execution semantics that Verilator follows are, it is currently unclear unless there is only a single clock.

gezalore, 2022-05-15:
With #3384 these attributes are no longer relevant for scheduling.

closed 2022-05-15 by gezalore