verilator/verilator #3188: compile error in V3Global.h

azimuth-tech, 2021-10-31:
The following error is encountered after typing 'make' as part of my own local install of Verilator. Is there an additional package that I need to install?

***********************************************
  In file included from ../V3Begin.cpp:31:0:                                                                                                                                                                                                                                             
../V3Global.h: In instantiation of ‘VRestorer<T>::VRestorer(T&) [with T = std::basic_string<char>]’:                                                                                                                                                                                   
../V3Begin.cpp:114:9:   required from here                                                                                                                                                                                                                                             
../V3Global.h:57:24: error: invalid initialization of non-const reference of type ‘std::basic_string<char>&’ from an rvalue of type ‘<brace-enclosed initializer list>’                                                                                                                
         , m_saved{permr} {}                                                                                                                                                                                                                                                           
                        ^                                                                                                                                                                                                                                                              
make[2]: *** [V3Begin.o] Error 1                                                                                                                                                                                                                                                       
make[2]: Leaving directory `/home/centos/src/project_data/verilator/src/obj_dbg'                                                                                                                                                                                                       
make[1]: *** [../bin/verilator_bin_dbg] Error 2                                                                                                                                                                                                                                        
make[1]: Leaving directory `/home/centos/src/project_data/verilator/src'                                                                                                                                                                                                               
make: *** [verilator_exe] Error 2     
***********************************************
Here is my OS version:
***********************************************
[centos@ip-172-31-55-69 verilator]$ cat /etc/os-release                                                                                                                                                                                                                                
NAME="CentOS Linux"                                                                                                                                                                                                                                                                    
VERSION="7 (Core)"                                                                                                                                                                                                                                                                     
ID="centos"                                                                                                                                                                                                                                                                            
ID_LIKE="rhel fedora"                                                                                                                                                                                                                                                                  
VERSION_ID="7"                                                                                                                                                                                                                                                                         
PRETTY_NAME="CentOS Linux 7 (Core)"                                                                                                                                                                                                                                                    
ANSI_COLOR="0;31"                                                                                                                                                                                                                                                                      
CPE_NAME="cpe:/o:centos:centos:7"                                                                                                                                                                                                                                                      
HOME_URL="https://www.centos.org/"                                                                                                                                                                                                                                                     
BUG_REPORT_URL="https://bugs.centos.org/"                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                                                       
CENTOS_MANTISBT_PROJECT="CentOS-7"                                                                                                                                                                                                                                                     
CENTOS_MANTISBT_PROJECT_VERSION="7"                                                                                                                                                                                                                                                    
REDHAT_SUPPORT_PRODUCT="centos"                                                                                                                                                                                                                                                        
REDHAT_SUPPORT_PRODUCT_VERSION="7"                                                                                                                                                                                                                                                     
************************************************
I just pulled from master before opening this issue. Here is the git sha I am using:
***********************************************
*master fdbf465 [origin/master] Fix array method names with parens (#3181) (#3183
***********************************************

Thank you for your time and your assistance.

wsnyder, 2021-10-31:
What is your "g++ --version"?  It looks like this is a different version in some respect so has some warning that newer gcc's have fixed.

If you use "make -k" are there other errors?

azimuth-tech, 2021-11-02:
Here is the g++ version:
*************************************************************************************
[centos@ip-172-31-55-69 verilator]$ g++ --version
g++ (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
Copyright (C) 2015 Free Software Foundation, Inc.

*************************************************************************************
Here are some more  errors from 'make -k':
*************************************************************************************
g++ -std=gnu++11  -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/centos/src/project_data/verilator\" -c V3Const__gen.cpp -o V3Const__gen.o
../V3Const.cpp:342:17: error: ‘unique_ptr’ is not a member of ‘std’
     std::vector<std::unique_ptr<VarInfo>> m_varInfos;  // VarInfo for each variable, [0] is nullptr
                 ^
../V3Const.cpp:342:17: error: ‘unique_ptr’ is not a member of ‘std’
../V3Const.cpp:342:33: error: template argument 1 is invalid
     std::vector<std::unique_ptr<VarInfo>> m_varInfos;  // VarInfo for each variable, [0] is nullptr
                                 ^
../V3Const.cpp:342:33: error: template argument 2 is invalid
../V3Const.cpp:342:40: error: expected unqualified-id before ‘>’ token
     std::vector<std::unique_ptr<VarInfo>> m_varInfos;  // VarInfo for each variable, [0] is nullptr
                                        ^
../V3Const.cpp: In constructor ‘ConstBitOpTreeVisitor::Restorer::Restorer(ConstBitOpTreeVisitor&)’:
../V3Const.cpp:168:29: error: invalid initialization of non-const reference of type ‘ConstBitOpTreeVisitor&’ from an rvalue of type ‘<brace-enclosed initializer list>’
             , m_restore{true} {}
                             ^
../V3Const.cpp: In member function ‘ConstBitOpTreeVisitor::VarInfo& ConstBitOpTreeVisitor::getVarInfo(const ConstBitOpTreeVisitor::LeafInfo&)’:
../V3Const.cpp:375:23: error: ‘m_varInfos’ was not declared in this scope
             baseIdx = m_varInfos.size();
                       ^
../V3Const.cpp:382:29: error: ‘m_varInfos’ was not declared in this scope
         VarInfo* varInfop = m_varInfos[idx].get();
                             ^
../V3Const.cpp: In constructor ‘ConstBitOpTreeVisitor::ConstBitOpTreeVisitor(AstNode*, unsigned int)’:
../V3Const.cpp:607:9: error: ‘m_varInfos’ was not declared in this scope
         m_varInfos.push_back(nullptr);
         ^
../V3Const.cpp: In static member function ‘static AstNode* ConstBitOpTreeVisitor::simplify(AstNode*, int, unsigned int, VDouble0&)’:
../V3Const.cpp:642:41: error: ‘class ConstBitOpTreeVisitor’ has no member named ‘m_varInfos’
         if (visitor.m_failed || visitor.m_varInfos.size() == 1) return nullptr;
                                         ^
../V3Const.cpp:652:32: error: ‘class ConstBitOpTreeVisitor’ has no member named ‘m_varInfos’
         termps.reserve(visitor.m_varInfos.size() - 1);
                                ^
../V3Const.cpp:657:33: error: ‘class ConstBitOpTreeVisitor’ has no member named ‘m_varInfos’
         for (auto&& v : visitor.m_varInfos) {

wsnyder, 2021-11-02:
While GCC 4.8.* is C++11, it's older than what we are currently able to automatically regress, so the Verilator code will need some minor cleanups.

Does your company require you to use this old centos/GCC 4.8 tree from 2013?  If not, using something more modern should "just work".  

Otherwise might you be willing to submit a patch to fix these issues, for example the V3Const seems mostly from needing an "include <memory>"?

azimuth-tech, 2021-11-03:
Hi Wilson, I am using one of the Amazon Web Services (AWS) servers which runs CentOS Linux. I had to manually install all the applications I am using. Yes, I can update to a later version of gcc, but I had try to do that before via:

> sudo yum update gcc

and it did not find any updates. Is there a way you recommend to getting the latest gcc version?

And yes, I would also be willing to submit a patch to fix these issues.

Thank you for your help and your work,
Kelvin

wsnyder, 2021-11-03:
I meant use a newer OS than CentOS 7, e.g. instead use Ubuntu 20.04 LTS.  RedHat keeps you at older GCC base versions when you update - an update will just get minor patches of GCC, not newer base versions.

>And yes, I would also be willing to submit a patch to fix these issues.

Would be appreciated, in that case feel free to stick with CentOS.  Thanks!

azimuth-tech, 2021-11-04:
Unfortunately the CentOS updates are managed by AWS and I don't think the user can update the OS. Some questions for you:

1. Can you specify in detail what the code that you want me to include is and which files it affects?
2. Do you want me to push to the verilator master branch?
3. Is there a regression I need to run to qualify the push?

Kelvin

azimuth-tech, 2021-11-08:
I didn't get a chance to debug further. It will take more time on my part to figure out the compiler warnings as I am unfamiliar with the code packages e.g V3Global.h. However, I noticed there is a pre-built Verilator download for version 3.922 for CentOS at https://centos.pkgs.org/7/epel-x86_64/verilator-3.922-1.el7.x86_64.rpm.html. I was able to run ventilator on CentOS with this install. Closing issue.

closed 2021-11-08 by azimuth-tech

wsnyder, 2021-11-08:
Great, just note that version is very old so is missing many features and fixes.  If it works for what you need, then great.