verilator/verilator #3189: Internal error of duplicate declaration when using generate blcok.

yanx21, 2021-11-01:
Hi
    I'm using the master branch of 'Verilator 4.215 devel rev v4.214-17-gfdbf465e'
    demo:
```verilog
module t (
);

parameter A = 0;
parameter B = 0;
parameter C = 0;

generate
    if (A == 1)
        integer i;
    else if (C == 1)
        integer i;
    else
        integer i;
endgenerate

generate
    if (B == 1)
        integer i;
    else if (C == 1)
        integer i;
    else
        integer i;
endgenerate

endmodule
```
```txt
%Error: t.v:24:17: Duplicate declaration of signal: 't.genblk1.i'
   24 |         integer i;
      |                 ^
        t.v:15:17: ... Location of original declaration
   15 |         integer i;
      |                 ^
%Error: Exiting due to 1 error(s)
```
I think this would be a problem cause `integer i` never actually duplicate in this demo.

I tried more tags and found this problem exist in tag v4.108 ~ v4.204, but not the previous ones.

yanx21, 2021-11-01:
There must be more than 3 `if-else branches`  in generate block to trigger this error.

wsnyder, 2021-11-01:
Thanks for the good report, yes, multiple genif's were busted, fixed I believe.

closed 2021-11-01 by wsnyder