verilator/verilator #3191: Is there a way to dependency inject modules in C++

oharboe, 2021-11-03:
Currently I implement a C++ harness that handles the top-level module signals. For instance, I can mock(emulate) an external SRAM in C++.

However, I would like to replace some of the lower level modules with C++ implementation too. These are lower level modules that have no top-level input/output signals. This can be useful to improve simulation speed, but also for debugging purposes.

My first instinct was to specify a top level module, then simply delete the Verilog modules that I want to implement in C++ in the hopes that Verilator would generate everything for the Verilog and that I would implement the rest in C++ myself.

If Verilator does not provide this option, then my next step would be to try to see if I can get Chisel to generate Verilog where the modules which I want to impelement in C++ have their signals exported to the top level module, in which case implementing the C++ version of those modules(mocking or emulating them in C++) is straightforward: just like emulating an external SRAM for instance.

wsnyder, 2021-11-03:
I presume you mean C++ modules under the Verilog top level (vs your C++ testbench top).

There is no option for this, as Verilator would still need to understand the clock domain of all output signals/etc. In theory --hierarchical is somewhat making a C++ module under a upper Verilog top, but this is not currently supporting any user C++, though I suppose could be improved to do so.

At present you'll therefore want to make a Verilog module that has calls into your C++ code - either using DPI routines (which would work on any simulator), or $c("...") (Verilator-specific and faster).

Closing as answered but feel free to followup!

closed 2021-11-03 by wsnyder

oharboe, 2021-11-03:
@wsnyder Thanks! In my case, this tells me that I have to go down the path of generating a top level Verilog module with the signals that I need for mocking the modules in question. There's a way to do this in Chisel that doesn't add too much cruft..