verilator/verilator #3193: No IData cast when shift items of array.

yanx21, 2021-11-06:
I'm using the master branch.
And here is the demo.
```systemverilog
module t(
    input logic vld,
    input logic [4:0] sft
);
logic vld_i[2];
logic [31:0] sft_vld[2] /*verilator public_flat_rw*/;

genvar i;
generate
  for (i=0; i<2; i=i+1) begin
    always @ (*) begin
      vld_i[i] = vld;
      sft_vld[i] = {vld_i[i], 31'h0} >> sft;
    end
  end
endgenerate

endmodule
```
Part of emited cpp code:
```cpp
vlSelf->t__DOT__sft_vld[0U] = ((vlSelf->t__DOT__vld_i[0U] << 0x1fU) >> (IData)(vlSelf->sft));
```
As we know, signed shift and unsigned shift are quit different. and `(vlSelf->t__DOT__vld_i[0U] << 0x1fU)` would be treat as a `signed int` by compiler, which is not my expectation.

However, If I use a simple logic signal rather than the array signal, It would be correct.
```systemverilog
// verilog demo
// ...
sft_vld = {vld_i, 31'h0} >> sft;
// ...
```
```cpp
// output cpp code
vlSelf->t__DOT__sft_vld = (((IData)(t__DOT__vld_i) << 0x1fU) >> (IData)(vlSelf->sft));
```
`t__DOT__vld_i` is cast to a IDate, and the shift would be a unsigned shift, which is correct.

wsnyder, 2021-11-07:
It's not related to signedness, but that the array dereference suppresses insertion of the cast.  I'm looking into a fix.

wsnyder, 2025-03-31:
This was fixed somewhere along the line, passes on master.