verilator/verilator #3194: Verilator not generating compile able system C code

amerhebi, 2021-11-08:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

What 'verilator --version' are you using? 4.034

  Did you try it with the git master version? -no

Would you be willing to try to fix Verilator yourself with assistance? yes

Since Verilator v 4.034 is already available in our conan remote repository 

I tried to install the above version and generate a systemc  for a simple verilog example blinky.v , with the below command 

verilator -Wall --sc --exe blinky.v

the generated systemc c code : Vblinky.cpp/*  does not compile correctly 

compile fails with the below error signature 

"Vblinky.cpp:(.text+0x0): multiple definition of `non-virtual thunk to Vblinky::~Vblinky()'; CMakeFiles/blinky_sc.dir/blinky_sc.cpp.o:blinky_sc.cpp:(.text+0x0): first defined here"

attached is the copy of the original code : blinky.v
generated code : Vblinky.cpp

Can you tell what setup issue I am missing that is preventing verialtor to generate compile able system C code ?

[vblinky.zip](https://github.com/verilator/verilator/files/7499096/vblinky.zip)
](url)

Many thanks for your help

wsnyder, 2021-11-08:
With --exe usually you put your top level on the command line - perhaps not doing that is causing this issue for you.   If you are proving the top level you didn't provide it in your zip file.

amerhebi, 2021-11-13:
Thank you Wilson Snyder for your feedback ,

   I tried to instantiate the blinky.v module in the top level sc_main.cpp (following the same steps as in the systemc Vour example in the Verilator website tutorial) and tried the below command to generate the system C code by invoking the top level in the command line as you have suggested previously :
verilator -Wall --sc --exe sc_main.cpp blinky.v 
Ref attached : vblinky.zip
[vblinky.zip](https://github.com/verilator/verilator/files/7531229/vblinky.zip)

But still the output code generated  vblinky/obj_dir/Vblinky.cpp is not compile able 

I have also tried to do the same verilator conversion steps with the exact same example referenced in the system C tutorial of the verilator website : https://verilator.org/guide/latest/example_sc.html#example-systemc-execution
command line : verilator -Wall --sc --exe sc_main.cpp our.v
Ref attached : 
[Vour.zip](https://github.com/verilator/verilator/files/7531240/Vour.zip)

But likewise the output code generated : Vour/obj_dir/Vour.cpp is not compile able 

This is still using the current version of verilator v4.034

Question : How is the output code being generated for the Vour example in the Verilator System C tutorial in your environment ? and what version of verilator are you using? 

I was wondering if the issue I am facing is due to an old version of verilator or a setup issue ?

Regards Abdallah

wsnyder, 2021-11-13:
You didn't include blinky_sc.h.  Please make a zip you've confirmed works standalone, thanks.

wsnyder, 2021-11-13:
[rewording]
You didn't include blinky_sc.h in the zip archive, so it doesn't compile. Please make a zip you've confirmed the zip has enough to show the exact problem described, thanks.

wsnyder, 2021-11-28:
Any update on this?  Did you get it working?

amerhebi, 2021-12-04:
[verilator-test2.zip](https://github.com/verilator/verilator/files/7653141/verilator-test2.zip)

Thank you Wilson Snyder for your feedback ,

  to make sure I have all the information required , I re-used the exact same example you provide in the verilator page :
https://verilator.org/guide/latest/example_dist.html
examples/cmake_hello_sc
Example building make_hello_sc with CMake.

The zipped directory has the exact setup I used 
In the CMakeLists.txt -> I specifically set the path of the Verilator installation 
set(ENV{VERILATOR_ROOT} ///////verilator-test2/build/.conan/data/verilator/4.034/_/_/package/356bcf71fd68612b83534850868d593984def918)
set(ENV{verilator_DIR} ///////verilator-test2/build/.conan/data/verilator/4.034/_/_/package/356bcf71fd68612b83534850868d593984def918)

Note : Above : /////// are relative paths I had to hide as instructed from my management. but you can still see the verilator installation directory  in the zipped directory 
verilator-test2/build/.conan/data/verilator/4.034/_/_/package/356bcf71fd68612b83534850868d593984def918/

The issue I am facing now is that when : 
- cd build 
-  cmake ..

I get the below error : verilator-test2/error.txt 

"- Check for working CXX compiler: /usr/intel/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
VERILATOR_ROOT is ///////verilator-t
est2/build/.conan/data/verilator/4.034/_/_/package/356bcf71fd68612b83534850868d5
93984def918
verilator_DIR is ///////verilator-te
st2/build/.conan/data/verilator/4.034/_/_/package/356bcf71fd68612b83534850868d59
3984def918
-- Could NOT find verilator (missing: verilator_DIR)
CMake Error at CMakeLists.txt:40 (message):
  Verilator was not found.  Either install it, or set the VERILATOR_ROOT
  environment variable"

amerhebi, 2021-12-04:
so the question I had above :  Why is the build flow not able to find the verilator package : verilator-test2/build/.conan/data/verilator/4.034///package/356bcf71fd68612b83534850868d593984def918/ as I have set the ENV variable in  CMakeLists.txt ? As shown above?

wsnyder, 2021-12-05:
This works fine for me, based on the error it looks like something on your end, you aren't pointing to the right verilator_DIR or VERILATOR_ROOT, or that directory doesn't contain a proper already-built Verilator kit.

amerhebi, 2021-12-11:
[make_hello_sc.zip](https://github.com/verilator/verilator/files/7696648/make_hello_sc.zip)

Thank you Wilson Snyder for your feedback ,

I will continue investigate the cmake recipe pointer issue to verilator_DIR/VERILATOR_ROOT.
Meanwhile to rule out the above setup issues  :

- I upgraded to verilator v4.214

- Ran the below steps manually (without cmake , but as instructed in the Verilator webpage : https://verilator.org/guide/latest/example_sc.html#example-systemc-execution)

- setenv  VERILATOR_ROOT amerhebi/Verilator/.conan/data/verilator/4.214/_/_/package/356bcf71fd68612b83534850868d593984def918
- setenv SYSTEMC_INCLUDE amerhebi/Verilator_conan/verilator-test1/build/.conan/data/systemc/2.3.3/syssim/stable/package/dc56540959bef913ccd30abdfafce169ed63dff9/include
- setenv SYSTEMC_LIBDIR amerhebi/Verilator_conan/verilator-test1/build/.conan/data/systemc/2.3.3/syssim/stable/package/dc56540959bef913ccd30abdfafce169ed63dff9/lib
- Then in the directory make_hello_sc/ (zipped and attached ) . I use the same example : Makefile  obj_dir  sc_main.cpp  top.v (From the Verilator webpage)
- generate the systemc C code with command : 
- verilator -Wall --sc --exe sc_main.cpp  top.v
- This generates the obj_dir with generated *.cpp *.mk files .
- Then I try to compile with the command  :  
- make -j -C obj_dir -f Vtop.mk Vtop
- but the compile fails with below error:

"sc_cor_qt::sc_cor_qt(bool)':
sc_cor_qt.cpp:(.text+0x2d6): undefined reference to `pthread_getattr_np'
//site/itools/em64t_SLES11/pkgs/gcc/9.1.0/.bin/../lib64/gcc/x86_64-suse-linux/9.1.0/../../../../x86_64-suse-linux/bin/ld: sc_cor_qt.cpp:(.text+0x2e6): undefined reference to `pthread_attr_getstack'
collect2: error: ld returned 1 exit status
make: *** [Vtop] Error 1
rm Vtop__ALL.verilator_deplist.tmp
make: Leaving directory `/////amerhebi/Verilator/make_hello_sc/obj_dir'"

detailed error : make_hello_sc/error.txt

Not sure why by following the same steps and same example as in the Verilator tutorial and latest v4.214, I am still not able to generate compile able systemc code ?

Can you share the same example "hello word" output from your end  (similar zipped directory : you can hide linux/unix directories)?  I am just curious to see whether obj_dir/Vtop.cpp output is different on your end ? Which causing it to compile in your case but not in my environment?

Many thanks for your help

wsnyder, 2021-12-11:
That is a link error, you're well past any problems caused by verilator-the-program and it generating a .cpp file.  Maybe the link command line needs -lqt?   Try verilating with "-LDFLAGS -lqd".  Also try making a systemc example that is completely outside verilator, it's possible that your distro's SystemC library is messed up.

wsnyder, 2022-01-01:
I'm presuming you figured this out, please feel free to followup otherwise if not.

closed 2022-01-01 by wsnyder

amerhebi, 2022-01-15:
[blinky.zip](https://github.com/verilator/verilator/files/7874085/blinky.zip)
[verilator-test2.zip](https://github.com/verilator/verilator/files/7874086/verilator-test2.zip)

Hi Wilson Snyder ,

 1) attached is a working example "blinky.zip" that is outside Verilator that compiles fine and I am able to run the systemc code sucessfully :
- cd blinky/build
- ./blinky_sc
0 s: led = 0
2047 ns: led = 1
4095 ns: led = 0
6143 ns: led = 1
8191 ns: led = 0

2) On the other hand if I use the same systemc make file of the above example but apply it to the example from the verilator website : https://verilator.org/guide/latest/example_sc.html#example-systemc-execution)   where the run directory for the latter is the attached verilator-test2.zip

- verilator -Wall --sc --exe sc_main.cpp top.v -LDFLAGS -lqd (as you have suggested)
- build generates the same below error ( Ref verilator-test2/Error.txt : for full details)

Vtop.cpp:(.text+0x110): multiple definition of `Vtop::~Vtop()'; CMakeFiles/Vtop.dir/Vtop.cpp.o:Vtop.cpp:(.text+0x110): first defined here
/nfs/site/itools/em64t_SLES11/pkgs/gcc/9.1.0/.bin/../lib64/gcc/x86_64-suse-linux/9.1.0/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/Vtop.dir/CMakeFiles/Vtop.dir/Vtop.dir/Vtop.cpp.o: in function `Vtop::Vtop(sc_core::sc_module_name)':Vtop.cpp:(.text+0x170): multiple definition of `Vtop::Vtop(sc_core::sc_module_name)'; CMakeFiles/Vtop.dir/Vtop.cpp.o:Vtop.cpp:(.text+0x170): first defined here
/nfs/site/itools/em64t_SLES11/pkgs/gcc/9.1.0/.bin/../lib64/gcc/x86_64-suse-linux/9.1.0/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/Vtop.dir/CMakeFiles/Vtop.dir/Vtop.dir/Vtop.cpp.o: in function `Vtop::Vtop(sc_core::sc_module_name)':
Vtop.cpp:(.text+0x170): multiple definition of `Vtop::Vtop(sc_core::sc_module_name)'; CMakeFiles/Vtop.dir/Vtop.cpp.o:Vtop.cpp:(.text+0x170): first defined here
/nfs/site/itools/em64t_SLES11/pkgs/gcc/9.1.0/.bin/../lib64/gcc/x86_64-suse-linux/9.1.0/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/Vtop.dir/CMakeFiles/Vtop.dir/Vtop.dir/Vtop.cpp.o: in function `Vtop::eval_step()':
Vtop.cpp:(.text+0x380): multiple definition of `Vtop::eval_step()'; CMakeFiles/Vtop.dir/Vtop.cpp.o:Vtop.cpp:(.text+0x380): first defined here
/nfs/site/itools/em64t_SLES11/pkgs/gcc/9.1.0/.bin/../lib64/gcc/x86_64-suse-linux/9.1.0/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/Vtop.dir/CMakeFiles/Vtop.dir/Vtop.dir/Vtop.cpp.o: in function `Vtop::final()':
Vtop.cpp:(.text+0x3e0): multiple definition of `Vtop::final()'; CMakeFiles/Vtop.dir/Vtop.cpp.o:Vtop.cpp:(.text+0x3e0): first defined here

So the systemc build flow outside verilator works fine on my end, but I am still not able to use that same build flow to compile systemc generated code from verilator.

and I also tried with -LDFLAGS -lqd as you have suggested previously.

is the code generated by verilator "verilator-test2/Vtop.cpp" as expected ? Do you get the same output for the same example ? Otherwise I am really not sure why the working systemc make file is not able to build successfully verilator generated systemc code. Let me know what are your other suggestions ?

It would be really helpful if you can attach the same Vtop example (same as verilator-test2/Vtop.cpp) from your end , with a successful compile log and the executable . This will enable me compare your working example of Vtop with my example of Vtop and see what is missing.

amerhebi, 2022-03-07:
Hi Wilson Snyder ,
I finally resolved the above issue , which was I was trying to compile the generated SystemC code . After finding out the only system c file to provide to the input is the wrapper file "testbench" and the other Verilog code will be automatically compiled after being translated in the build sub directory.