verilator/verilator #3195: Not able to build verilator when downloading latest version 4.214 from master branch

amerhebi, 2021-11-08:
Thanks for taking the time to report this.

Can you attach an example that shows the issue? yes  (Must be openly licensed, ideally in test_regress format.)

What 'verilator --version' are you using?  trying to download and build version 4.214.
 Did you try it with the git master version? yes

Would you be willing to try to fix Verilator yourself with assistance? yes

After running the below steps to install latest version of verilator v4.214 
1) git clone https://github.com/verilator/verilator
2) unsetenv VERILATOR_ROOT  #
3) cd verilator
4)  git pull         # Make sure git repository is up-to-date
5)  git tag          # See what versions exist
6) autoconf         # Create ./configure script
7) ./configure      # Configure and create Makefile
8) make -j `nproc`  # Build Verilator itself (if error, try just 'make')

Note for step 8 tried both make -j `nproc` and make 

but the build fails with the below error signature 

"In file included from ../V3Ast.h:2937:0,
                 from ../Verilator.cpp:18:
../V3AstNodes.h:2631:13: error: looser throw specifier for ‘virtual AstPackageImport::~AstPackageImport()’
In file included from ../Verilator.cpp:18:0:
../V3Ast.h:1488:13: error:   overriding ‘virtual AstNode::~AstNode() noexcept (true)’
In file included from ../V3Ast.h:2937:0,
                 from ../Verilator.cpp:18:
../V3AstNodes.h:2650:13: error: looser throw specifier for ‘virtual AstIface::~AstIface()’
In file included from ../Verilator.cpp:18:0:
../V3Ast.h:2855:13: error:   overriding ‘virtual AstNodeModule::~AstNodeModule() noexcept (true)’
In file included from ../V3Ast.h:2937:0,
                 from ../Verilator.cpp:18:
../V3AstNodes.h:2676:13: error: looser throw specifier for ‘virtual AstMemberSel::~AstMemberSel()’
In file included from ../Verilator.cpp:18:0:
../V3Ast.h:1869:13: error:   overriding ‘virtual AstNodeMath::~AstNodeMath() noexcept (true)’
In file included from ../V3Ast.h:2937:0,
                 from ../Verilator.cpp:18:
../V3AstNodes.h:2715:13: error: looser throw specifier for ‘virtual AstModportFT"

Let me know what could be the setup issue that would prevent latest version of verilator to build correctly ?

Many thanks for your help.

wsnyder, 2021-11-08:
I haven't seen this error before.  A google of "error: looser throw specifier" suggests it is related to noexcept, which Verilator doesn't use.  What distro and distro version are you using?  Can you try googling the errors and trying fixes in the C++ code?

wsnyder, 2021-11-28:
Any update on this?

wsnyder, 2021-12-12:
I'm assuming you got it sorted.

closed 2021-12-12 by wsnyder