verilator/verilator #3197: 4.215 gives simulation mismatch

agrobman, 2021-11-11:
@wsnyder , 

We built newest 4.215  version and it gives wrong results  for a FPU instruction in simulation of our CPU core.
I've traced this to that equation:
[image]
this is a snapshot from xcelium,
verilator sets i2s_round_inc_x1  signal to 0 , all  RHS signals in verilator  match  Xcelium signals ... I don't see anything special in this equation. 
BTW, only our FPU tests fail out of our regression tests suite. (they pass with 4.200)

I'm not sure how to debug this problem further , please, advice.
Meanwhile we will try 4.214

wsnyder, 2021-11-11:
Does -Oo help?

agrobman, 2021-11-11:
we use -Os ..

wsnyder, 2021-11-11:
I'm referring to a Verilator flag, this might work around your issue.

agrobman, 2021-11-11:
the failing tests pass with -Oo ... what are consequences of the -Oo option on simulations ?

wsnyder, 2021-11-11:
-Os disables a new ConstBitOpTree optimization added a few releases ago.  @yTakatsukasa mind taking a look?

agrobman, 2021-11-11:
Whilson, thanks

We compile our stuff with following commands:

```
verilator  -cc -CFLAGS -std=c++11 -Wno-WIDTH  -Wno-MULTIDRIVEN --autoflush  -Oo -f flist -v gitstate.v +define+USE_GIT_STATE --top-module tb_top -exe test_tb_top.cpp 
 make -j -C obj_dir/ -f Vtb_top.mk OPT_FAST="-Os"
```

Is there  better, more general way  to do this? I don't like that we use two commands / some verilator "internals"(make script) , which may change one day..

agrobman, 2021-11-11:
Is there any performance degradation with -Oo?

wsnyder, 2021-11-11:
Add "-j --build --MAKEFLAGS OPT_FAST=-Os" and Verilator will call make for you.

wsnyder, 2021-11-11:
"-Oo" is probably a few percent on most designs.

agrobman, 2021-11-11:
is this -Oo temporary measure to neutralize ' our OPT_FAST=-Os effect ? and Do we really need to supply this OPT_FAST or by default verilator will select reasonable C++ optimization compile option?

also (why) do we need this -CFLAGS -std=c++11 option?

wsnyder, 2021-11-11:
-Oo is a verilator flag and completely unrelated to GCC's -Os.

You only need the -std=c++11 if you are using a (as of 2021) 10 year old compiler that doesn't default to C++11.  If you remove it and everything compiles you're good.

agrobman, 2021-11-11:
one more question. How to suppress make output from STDOUT with the --build verilator option?

agrobman, 2021-11-11:
hi, 
Please, let me know, since what version was this optimization  introduced and when (if) will the issue be solved ? ..

and how do I get description of all -O options?

agrobman, 2021-11-12:
4.214 release has the same issue

yTakatsukasa, 2021-11-13:
Here is the history of the optimization.
- v4.110
bb5384bc5031eebc874cf26a83b671345036b33a initially introduced
- v4.200
7469166c38a50a0d38afb92a8146e69db5e32ae3 Fix
- v4.202
56dd58d259e8d90ebcfb3ee44af494945d92c8f0 Assertion failure
- v4.204
4351abfe7107487a524fe51b631119487c2fa6b9 Assertion failure
8624ce6a84dcd0908a9d721420e1f790ffaf4ba1 Better opt
- v4.210
ec4eb18846cf6fbe7edf7ee886ee3355df33ebfb Fix
- v4.212
4ec8bc358996ab7dddc89244e7d99f54e14a802b Fix
c69ddc46f87ac3424fe3770e6fec5c751cf972a5 Fix
8681861be9aa167cecbb267b10881dc8d87e1b83 Improvement and many rewrite
- master
b08c694cd626e349f4bad8048ab946485c5ba6b0 Fix

I am wondering which exact revision are you referring as `newest 4.215`. Does it include b08c694cd626e349f4bad8048ab946485c5ba6b0 ?
According to your experiment, the bug was introduced between 4.200 and 4.214.

Could you also try the following revisions ?
- The latest master that includes b08c694cd626e349f4bad8048ab946485c5ba6b0
- v4.212
- v4.210
- v4.204
- v4.202

If you can share your design, ideally shrunk to small enough, I expect it will not take so long.

agrobman, 2021-11-13:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

4.215 was built from master day before my issue was filed.
I'm not sure I can share design with you. It is not for open source.

I will try to ask our IT to build older versions next week.

bb5384b<https://github.com/verilator/verilator/commit/bb5384bc5031eebc874cf26a83b671345036b33a> initially introduced
  *   v4.200
7469166<https://github.com/verilator/verilator/commit/7469166c38a50a0d38afb92a8146e69db5e32ae3> Fix
  *   v4.202
56dd58d<https://github.com/verilator/verilator/commit/56dd58d259e8d90ebcfb3ee44af494945d92c8f0> Assertion failure
  *   v4.204
4351abf<https://github.com/verilator/verilator/commit/4351abfe7107487a524fe51b631119487c2fa6b9> Assertion failure
8624ce6<https://github.com/verilator/verilator/commit/8624ce6a84dcd0908a9d721420e1f790ffaf4ba1> Better opt
  *   v4.210
ec4eb18<https://github.com/verilator/verilator/commit/ec4eb18846cf6fbe7edf7ee886ee3355df33ebfb> Fix
  *   v4.212
4ec8bc3<https://github.com/verilator/verilator/commit/4ec8bc358996ab7dddc89244e7d99f54e14a802b> Fix
c69ddc4<https://github.com/verilator/verilator/commit/c69ddc46f87ac3424fe3770e6fec5c751cf972a5> Fix
8681861<https://github.com/verilator/verilator/commit/8681861be9aa167cecbb267b10881dc8d87e1b83> Improvement and many rewrite
  *   master
b08c694<https://github.com/verilator/verilator/commit/b08c694cd626e349f4bad8048ab946485c5ba6b0> Fix

agrobman, 2021-11-18:
4.204 is 'good'

agrobman, 2021-11-18:
4.210 is 'good' too

agrobman, 2021-11-18:
4.212 is 'bad'

yTakatsukasa, 2021-11-19:
Thanks for the experiment.
We can just focus on commits between v4.210 and v4.212.

I'd like to narrow down further.

1. Could you try the following 2 versions ?
  - fffc63970de97298454dcd0eb3fa451753a23399 (The previous commit of 8681861)
  - 8681861be9aa167cecbb267b10881dc8d87e1b83

2. Could you run `verilator --version` for v4.215 that you tried ? (The version message contains git hash)
  > 4.215 was built from master day before my issue was filed.

agrobman, 2021-11-19:
% verilator --version
Verilator 4.215 devel rev v4.214-28-g185e5d8

agrobman, 2021-11-23:
fffc639 - every test fails (our cpu runs only 2 first instructions) - this commit probably broken completely ...
8681861 - fails the same way as 4.215

gezalore, 2021-11-23:
> [fffc639](https://github.com/verilator/verilator/commit/fffc63970de97298454dcd0eb3fa451753a23399) - every test fails (our cpu runs only 2 first instructions) - this commit probably broken completely ... [8681861]

I mean, this passes the whole test suite, including the extended test suite that includes EH1, EH2 and EL2. Also, for the record, I am not sure you are aware of this, and I hope it is not intentional, but I often find the style of your comments somewhat rude. We all have different execution environments, so what is "completely broken" on your system, might be completely fine on another. Additionally, the trailing ellipses (...) generally don't convey any information other than signifying annoyance, and in the part of the world I am from they can be misconstrued as offensive.

agrobman, 2021-11-24:
@gezalore , Sorry If I offended you, but what I meant is that if for all other releases we had only a few FPU tests failing (what points to some specific structures, effected by this optimizations),   all our tests were failing with this particular commit. (something more generic did not work right).  I provided this info as a possible hint to help find the problem. 

The design we are working on is quite different from published SweRVs , (although  the new design style should be very similar to the SweRVs).  
I understand that you can't have  all possible code samples to test the tool, so such issues may happen. I personally  spent quite a lot of time trying to isolate this problem and also asked other pretty busy people in the company to help me with this. 
BTW, I tried to build a small example, including "failed" logic  but it was working correct with all released versions.

I thought its our common goal to make the tool better. Also, please, consider possible cultural differences and not be offended if somebody is not aware about your sensitivities. 

Let's be more constructive  in how to find and fix root source of this problem. 
this is really hard thing - some code works, some does not, meaning that people can hit the case by just slightly modifying their code and then spending a lot of time trying to chase a "ghost".

agrobman, 2021-11-24:
Any thoughts about next steps in finding root cause of the issue?

yTakatsukasa, 2021-11-26:
Here is what I understand. Correct me if I misunderstand.
The bug was introduced between v4.210 and v4.212. 
This issue is different from #3182 as 185e5d8 that is newer than b08c694cd626e349f4bad8048ab946485c5ba6b0 still show a wrong result.

I am still being confused by the result that fffc639 is worse than other commits.
I expected the commit to pass or fail just as other commits.

Anyway I will reread V3Const.cpp as your design is proprietary.

If you can make a small example that can be shared with us, that would be wonderful.

If all signals of the original code are FFs, then I believe bit op tree can optimize correctly.
But there must be other circuits from FFs such as bit-select/and/or/not/shift and I guess such circuits triggered the bug.
So if you try to make an example, please keep the original registers.

e.g.

Assume the original design looks as below,
```systemverilog
   reg [63:0] i2s_xl;
   wire i2s_lsb_xl;
   assign i2s_lsb_xl = i2s_xl[0];
```
You cannot simplify as below.
```systemverilog
   reg i2s_lsb_xl;
```

agrobman, 2021-11-26:
I was able to build a small example :
run is compile command , 
to run the test:
obj_dir/Vtb_top

It fails as

```
VerilatorTB: Start of sim

mismatch fcvt_flags_fx1 Exp=01 Act=00
- tb_top.sv:67: Verilog $finish

VerilatorTB: End of sim
```
About this example : I recorded inputs and outputs of our rtl module during xcelium simulation  to fpu_dump file.
this example loads this file to a verilog memory and replays it driving the module inputs from the memory and compares output every clock. 

[issue3197.tar.gz](https://github.com/verilator/verilator/files/7610256/issue3197.tar.gz)

closed 2021-11-30 by yTakatsukasa

yTakatsukasa, 2021-11-30:
Fix (#3221) is merged to master.
Please try the latest master. If you still hit the issue, please reopen.

PR #3221 (2021-11-30): Fix bit op tree optimization
I hope this PR fixes #3197 

When AST as below is passed to bit op tree optimizer,
```
 OR 0x55c24b849ad0 <e61271#> {d393cw} @dt=0x55c24b941b70@(G/wu32/1)
1: CCAST 0x55c24bb31470 <e95178#> {d277ca} @dt=0x55c24b941b70@(G/wu32/1) sz32
1:1: SHIFTR 0x55c24ba5f790 <e95175#> {d277ca} @dt=0x55c24b841df0@(G/w64)
1:1:1: VARREF 0x55c24b89a820 <e95166#> {d277bp} @dt=0x55c24b841df0@(G/w64)  var [RV] <- VAR 0x55c24b8dd9f0 <e56393#> {d78bc} @dt=0x55c24b841df0@(G/w64)  var [FUNC] VAR
1:1:2: CONST 0x55c24b9ca020 <e95167#> {d277cb} @dt=0x55c24b843c20@(G/swu32/6)  6'h27
2: NEQ 0x55c [... cut]
commit 1160dddc (2021-11-30): Fix bit op tree optimization (#3221)

* Add a test to reproduce #3197

* Fix #3197. Optimize correctly even if a variable is >32

* Quick exit instead of continue. No functional change is intended.

* Delete comment-out line.

* update per review comment