verilator/verilator #3198: import-DPI with an open array for the packed dimension

quark17, 2021-11-12:
The SV DPI spec mentions "open arrays" and give `bit[]` as a legal example.  I tried writing this declaration (where the function can work for multiple sizes, with `arg1` telling it what size it was called with):
```
import "DPI-C" function void poly_action(output bit[] res, bit[] arg0, int arg1);
```
and I see that Verilator 4.200 (2021-03-12 rev v4.110-43-g5022e81af) rejects the syntax:
```
%Error: dpi_declarations.v:1:53: syntax error, unexpected ']', expecting TYPE-IDENTIFIER
```
I tried looking in `verilog.y` to see why it didn't parse, and I can't see why it's not -- maybe that's a bug?  I do see this, which suggests that it should give a nicer message about not being supported:
```
packed_dimension<nodeRangep>:   // ==IEEE: packed_dimension                                                                                  
                anyrange                                { $$ = $1; }
        |       '[' ']'
                        { $$ = nullptr; BBUNSUP($<fl>1, "Unsupported: [] dimensions"); }
        ;
```
FYI, here's the path that I followed: the first option for `dpi_import_export` expects `function_prototype`; the first arm of that expects `tf_port_listE` between the parentheses; and that's `tf_port_listList`, which is a comma separated list of `tf_port_item`; the second arm of that is `tf_port_itemFront` followed by the name; and one of the arms of that is `tf_port_itemDir data_type` (a port direction like `output` followed by a data type); and that supports a type followed by `packed_dimensionListE`; which includes `packed_dimensionList`; which includes `packed_dimension`.

Separate from whether the error is being reported properly, would it be possible to support `bit[]` in DPI-C?

wsnyder, 2021-11-12:
If you'd like to make a pull request for "bit[]" that would be welcome.  I suspect it won't get much attention for a long time otherwise.  To support this we'd first need some tests that work on other simulators, and check at least the widths that cross around the important Verilator widths of 1,7,8,15,16,31,32,63,64,65.  Then you'd need to fix this parser issue, and update the appropriate code in V3Task to support the open range.

To understand the grammar, run Verilator with "--debug --debugi-bison 9" and you'll see the rules called.

```
Next token is token ']' ()
%Error: ....: syntax error, unexpected ']', expecting TYPE-IDENTIFIER
Error: popping nterm packageClassScopeE ()
Error: popping token '[' ()
Error: popping nterm signingE ()
Error: popping nterm integer_vector_type ()
Error: popping nterm tf_port_itemDir ()
Error: popping nterm $@22 ()
Error: popping token '(' ()
Error: popping nterm funcId ()
Error: popping token "function" ()
Error: popping nterm dpi_importLabelE ()
Error: popping nterm dpi_tf_import_propertyE ()
Error: popping token "STRING" ()
Error: popping token "import" ()
```

Look at src/obj_dbg/V3ParseBison.output to understand the stack numbers.

quark17, 2021-11-23:
Thanks!  I should be able to attach an example that works with Questa/Modelsim.  I understand that this is not a priority, and I myself may not have time to look at submitting a PR for some time.

quark17, 2021-12-02:
FYI, I notice that the copy of `svdpi.h` in the Verilator repo has a `*` in the declaration of `svPutPartSelectLogic` that doesn't appear to be there in the spec:
```
  421c430
  < XXTERN void svPutPartSelectLogic(svLogicPackedArrRef d, const svLogicVec32 s,
  ---
  > XXTERN void svPutPartSelectLogic(svLogicPackedArrRef d, const svLogicVec32* s,
```
This is in the deprecated part of the API, which Verilator doesn't implement, so I assume it hasn't mattered.  Is it a typo?  (Should Verilator just remove the deprecated declarations from its `svdpi.h`?)

Also, I have made two examples that work in QuestaSim, one using `bit[]` and one using `logic[]`: [dpi_open_array_packed.tar.gz](https://github.com/verilator/verilator/files/7639183/dpi_open_array_packed.tar.gz)

They may not exhaustively test the `svdpi.h` API, but they touch a few.  The imported function takes two open packed arrays of the same length and adds them, returning a packed array of the same length.  The test also natively adds the packed arrays, and tests whether the DPI function computed the same value.  The output from QuestaSim in both tests is:
```
0 + 1 = 1 (1)
1001001 + 1001001 = 0010010 (1)
01010010 + 10010001 = 11100011 (1)
100100010100010 + 001001001010101 = 101101011110111 (1)
1010010110000001 + 1101100001001100 = 0111110111001101 (1)
0001111000011111111000011110000 + 0001000100000010000111001000101 = 0010111100100001111111100110101 (1)
10101010010101011000000101000010 + 01010101101010100010010010000001 = 11111111111111111010010111000011 (1)
111000011110000111100001111000011110000111100001111000011110000 + 001111000001101000010110000011100001001000001100000101000000101 = 000111011111101111110111111011111111001111101101111101011110101 (1)
1010101001010101100000010100001010101010010101011000000101000010 + 0101010110101010001001001000000101010101101010100010010010000001 = 1111111111111111101001011100001111111111111111111010010111000011 (1)
11010101001010101100000010100001010101010010101011000000101000010 + 10011010101011010101000100100100000010101010110101010001001001000 = 01101111110110000001000111000101010111111101100000010001110001010 (1)
ERROR: poly_add: size mismatch: 1, 7, 7
0
```
The makefiles will build and run QuestaSim by default.  You can run `make SIM=verilator` to use Verilator.  It results in the parser error:
```
%Error: mod.v:2:51: syntax error, unexpected ']', expecting TYPE-IDENTIFIER
    2 | import "DPI-C" function void poly_add(output bit [], input bit [], input bit []);
      |                                                   ^
```
```
%Error: mod.v:2:53: syntax error, unexpected ']', expecting TYPE-IDENTIFIER
    2 | import "DPI-C" function void poly_add(output logic [], input logic [], input logic []);
      |                                                     ^
```

quark17, 2021-12-02:
The `ERROR` line is expected.  The imported function tests whether the sizes of the input and output arrays are all the same, prints that message if not, and returns all zeros; the following line of the output is the module printing the value that it received, which is `0`.

wsnyder, 2021-12-02:
As to putPartSelectLogic, svLogicVec32 is a structure so in C (which these externs are) it is not legal to pass a structure by value, therefore this must be a pointer.   I would prefer the header continue to follow IEEE as close as is possible (e.g. compiles without warning) versus e.g. commenting out deprecated definitions.

quark17, 2021-12-09:
I believe the parser bug is this in Verilator:
```
data_typeBasic<nodeDTypep>:
    integer_vector_type signingE rangeListE { ... }
  | ...
```
where `rangeListE` does not allow unsized dimensions (`[]`), while the IEEE 1800-2017 spec has this:
```
data_type ::=
    integer_vector_type [ signing ] { packed_dimension }
  | ...
```
where `packed_dimension` is allowed to be an unsized dimension.

I believe that the parsing is fixed by changing `rangeListE` to be `packed_dimensionListE`.

When I make that change, the previously attached example then gives this error:
```
%Error-UNSUPPORTED: mod.v:2:50: Unsupported: [] dimensions
    2 | import "DPI-C" function void poly_add(output bit [] a, input bit [] b, input bit [] c);
      |                                                  ^
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=4.215
%Error-UNSUPPORTED: mod.v:2:66: Unsupported: [] dimensions
    2 | import "DPI-C" function void poly_add(output bit [] a, input bit [] b, input bit [] c);
      |                                                                  ^
%Error-UNSUPPORTED: mod.v:2:82: Unsupported: [] dimensions
    2 | import "DPI-C" function void poly_add(output bit [] a, input bit [] b, input bit [] c);
      |                                                                                  ^
%Error: Exiting due to 3 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

And that seems to be what we expect, from this code mentioned earlier:
```
packed_dimension<nodeRangep>:
    anyrange  { $$ = $1; }
  | '[' ']'   { $$ = nullptr; BBUNSUP($<fl>1, "Unsupported: [] dimensions"); }
  ;
```

I would submit a PR for this small fix, but I can't check that it passes the testsuite because the testsuite doesn't pass on a fresh clone for me -- I'm still figuring that out.  Although I guess if I push a commit to my own fork, the CI will run the testsuite for me?

I do have two questions, though.  One, I notice that `rangeList` looks like this:
```
rangeList<nodeRangep>:          // IEEE: {packed_dimension}
    anyrange                                { $$ = $1; }
  | rangeList anyrange                      { $$ = $1; $1->addNext($2); }
  ;
```
while `packed_dimensionList` looks like this:
```
packed_dimensionList<nodeRangep>:       // IEEE: { packed_dimension }
    packed_dimension                        { $$ = $1; }
  | packed_dimensionList packed_dimension   { $$ = VN_CAST($1->addNext($2), NodeRange); }
  ;
```
Why is `packed_dimensionList` using `VN_CAST`?

Two, I see that one user of `packed_dimensionList` is passing the value to `createArray` -- see this arm of `data_type`:
```
packageClassScopeE idType packed_dimensionListE
  { AstRefDType* const refp = new AstRefDType{$<fl>2, *$2, $1, nullptr};
    $$ = GRAMMARP->createArray(refp, $3, true); }
```
For the `integer_vector_type`, the dimensions are passed to `addRange`:
```
integer_vector_type signingE packed_dimensionListE
  { $1->setSignedState($2); $$ = GRAMMARP->addRange($1,$3,true); }
```
What is the difference?

closed 2026-10-18 by wsnyder