verilator/verilator #3199: Infinite loop in Verilator --lint-only (version 4.200_1)

jrrk2, 2021-11-13:
In the statement: parameter type sz=sz, the definition recurses endlessly whereas the expected behaviour is to import the definition of sz from the imported package, or give a suitable error message.

package types;

  parameter [7:0] wid = 8;

   typedef logic [wid-1:0] sz;

endpackage // types

module count8 
import types::*;
 # (parameter type sz = sz) (input clk, input rst, input en, output sz q);

   always @(posedge clk or posedge rst)
     if (rst)
       q = {{types::wid}{1'b0}};
     else if (en)
       begin
	  q = q + 1;
       end

endmodule

wsnyder, 2021-11-22:
As properly importing this is related to the parser limitation of #2412, so is unlikely to be fixed quickly as that is a complicated issue.  As you note a workaround is to use the package reference.

Of course without that Verilator should not be hanging.  It seems to recurse in an internal dtypeNoRef function call, which makes it a little more complicated then I initially had hoped, but will get to it.

jrrk2, 2021-11-22:
Thanks for the update, I thought I should bring it to your attention as it was a pathological case for a tool which is generally robust.

closed 2021-11-23 by wsnyder