verilator/verilator #3200: Could verilator support visible signals in protect lib?

EDAHUB, 2021-11-14:
Hi,
Currently we try verilator on big design and use hierarchical mode, we can't dump trace in hierarchical blocks. It is hard for us to debug. Is it possible to support dump signals in protect lib?

udif, 2021-11-14:
Verilator's support of protected verilog code only includes knowing how to gracefully skip protected content.
If your protected library exposes the module's interface before diving into encrypted mode then you can see those signals.
You cannot get into the protected sections, or simulate them with Verilator.

EDAHUB, 2021-11-14:
then maybe we should raise this as a request for hierarchy mode. Hierarchy is for solve the long compile issue. The design including the submodules are still under debug so visibility inside is better to be reserved.

wsnyder, 2021-11-14:
To work towards this and generally reduce confusion I'm adding --lib-create which is basically --protect-lib without the protection aspects.

wsnyder, 2021-11-14:
I have manually-edited Verilated code able to hierarchically trace. Still need to figure out how to plumb in the context passing.

closed 2021-11-27 by wsnyder