verilator/verilator #3201: Build error on Centos7  - verilator v4.214

mmatzev, 2021-11-14:
I try to build version 4.214 on centos7

Error message during make -j 4:
g++ -std=gnu++11  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/iss_ki_predict/Design/KIPRED01/_cad/digital/workareas/vkr/verilator_4214/share/verilator\" -c ../V3AstNodes.cpp -o V3AstNodes.o
In file included from ../V3Begin.cpp:31:0:
../V3Global.h: In instantiation of 'VRestorer<T>::VRestorer(T&) [with T = std::basic_string<char>]':
../V3Begin.cpp:114:9:   required from here
../V3Global.h:57:24: error: invalid initialization of non-const reference of type 'std::basic_string<char>&' from an rvalue of type '<brace-enclosed initializer list>'
         , m_saved{permr} {}
                        ^
make[2]: *** [V3Begin.o] Error 1
make[2]: *** Waiting for unfinished jobs....

used g++:
ix1357 [verilator] $ g++ --version
g++ (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mmatzev, 2021-11-14:
Small note: 
- Same error with 4.212
- Compile works with 4.200

wsnyder, 2021-11-14:
Can you use CentOS8? That is known to work - centos7 is over 10 years old so is not supported as part of our typical regressions.

Alternatively please consider providing a pull request that fixes the compiler issues.

wsnyder, 2021-11-14:
Correction, CentOS Stream 8.  The regular CentOS 8 is near end-of-life.

mytbk, 2021-11-15:
If you still want CentOS 7, you can build and install the latest binutils, GCC, GNU make on the system and use them.
According to docs/guide/deprecations.rst, verilator will require C++14 in the near future.

fredrequin, 2021-11-17:
Under CentOS 7, you must install the Software Collection. Then, you can enable GCC 9 with the command : `slc enable devtoolset-9 bash`.

mmatzev, 2021-11-19:
Works with scl enable devtoolset-9 bash

closed 2021-11-19 by mmatzev

max-elastics, 2022-04-20:
CentOS 7 is still very widely required in the EDA industry, where Verilator is used in tandem.   Please consider keeping support for it. 

From the hints above I was able to compile 4.220 using 
``` 
sudo yum install centos-release-scl
sudo yum install devtoolset-7
scl enable devtoolset-7 bash
./configure --prefix=/destination/path/to/verilator/verilator-4.220
make -j4 install
```