verilator/verilator #3202: Verilator thinks code contains delayed assignment for large arrays

secworks, 2021-11-16:
I'm running Verilator as a linter.

In my code I have a state array with 256 bytes declared as:
  reg [7 : 0]  state_iaddr;

The state array is reset using a for loop:

  always @ (posedge clk)
    begin: reg_update
      integer i;

      if (!reset_n) begin
        for (i = 0 ; i < 256 ; i = i + 1) begin
          state[i] <= 8'h0;
        end
...
...

But when linting I get the following error message:

verilator +1364-2001ext+ --lint-only -Wall -Wno-fatal -Wno-DECLFILENAME ../src/rtl/rc4_core.v
%Error-BLKLOOPINIT: ../src/rtl/rc4_core.v:123:20: Unsupported: Delayed assignment to array inside for loops (non-delayed is ok - see docs)
  123 |           state[i] <= 8'h0;
      |                    ^~

By varying the loop exit condition (i < X), I can see that when the X is 64 or smaller, Verilator does not respond with the error. But if the loop exit condition is 65 and larger (up to the array size), I get the error message above. I believe that this is an incorrect response from Verilator.

I'm using 'verilator --version': "Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)"

secworks, 2021-11-16:
The code (BSD 2-clause licensed) is available here:

https://github.com/secworks/rc4/blob/restart/src/rtl/rc4_core.v

The easiest is probably to check out the repo, switch to the restart branch and run make lint in the toolruns subdir.

gezalore, 2021-11-16:
Generally Verilator does not support delayed assignments to arrays inside loop statements (as the message says). However, short loops are unrolled when, and you probably discovered the default unroll limit. See https://verilator.org/guide/latest/exe_verilator.html?highlight=unroll#cmdoption-unroll-count on how to increase this. Setting the unroll limit to 256 in your case should make your code work.

secworks, 2021-11-16:
Ah, cool! Thanks for a prompt response. I've added the loop count limit as a lint flag.

closed 2021-11-16 by secworks