verilator/verilator #3204: Signed-ness ignored in display of value without format spec

quark17, 2021-11-16:
I notice that Verilator has different output for the following two statements:
```
$display("%d", $signed(32'hAAAAAAAA));
$display($signed(32'hAAAAAAAA));
```
```
-1431655766
2863311530
```
The second statement has an argument to `$display` without a corresponding format specification.  The SystemVerilog 2017 spec seems to say that this is illegal (for types other than string or an array of bytes).  The Verilog 2005 spec says that such arguments are displayed with the default decimal format.  I would assume that means `%d`, which is why I compared the second statement with the first statement.

I don't know what behavior is intended for Verilator, but I would assume that if the second line is accepted, then the value should be the same as the first?  The Questa and Icarus simulators give the same value (`-1431655766`) in both cases.

I am using `Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)`.  (I wanted to test with the HEAD, but I am having trouble building a working Verilator on my mac.)

quark17, 2021-11-16:
I would imagine you have boilerplate to put the display statements into, but here is a self-contained example with Makefile: [verilator_issue_3204.tar.gz](https://github.com/verilator/verilator/files/7550628/verilator_issue_3204.tar.gz)

It includes display statements of the unsigned values, and the output looks like this:
```
$ ./display_signed.exe 
-1431655766
2863311530
2863311530
2863311530
- display_signed.v:12: Verilog $finish
```

wsnyder, 2021-11-17:
Adding another case to avoid breaking unsigned:

```
      $display("%d", $signed(32'hAAAAAAAA));  // -1431655766
      $display($signed(32'hAAAAAAAA));  // -1431655766
      $display("%d", $unsigned(-2));  // 4294967294
      $display($unsigned(-2));  // 4294967294
      $display("%d", 32'hAAAAAAAA);  // 2863311530
      $display(32'hAAAAAAAA);  // 2863311530
```

Something's wrong in handing default arguments, probably in V3Width which does this.  Perhaps you'd be able to take a look?

The test cases would go into test_regress/t/t_display.v, then update the golden output with "test_regress/t/t_display.pl --gold"

wsnyder, 2021-11-17:
Looked into and simple fix.

closed 2021-11-17 by wsnyder