verilator/verilator #3205: Error when readmem file does not end at the final address (regardless of whether the file has address specifications)

quark17, 2021-11-17:
I have this statement in a Verilog module:
```
$readmemh("arr.hex", arr, 0, 7);
```
And I have a file `arr.hex` that contains address specifications:
```
// 8 x 8-bit memory contents
00
11
22

// skip entry 3
@4
44
55
66
77

// return to entry 3
@3
33
```
When the system task executes, Verilator aborts with this error:
```
%Error: foo.hex:16: $readmem file ended before specified final address (IEEE 2017 21.4)
```

In Section 21.4 of the SV 2017 spec, I see this:
> A warning shall be issued if the number of data words in the file differs from the number of words in the range implied by the start through finish addresses and no address specifications appear within the data file. In this case, the data words that are contained in the file shall be loaded into the memory beginning at the start address, and memory addresses for which the file does not contain data words are not modified by the operation.

This is about files where "no address specifications appear within the data file", so it does not apply to my file.  But I am unable to find any other part of Section 21.4 that deals with the final address.  Is there another paragraph that I'm missing?

If this is the cited paragraph: Is it intentional that Verilator treats this as a fatal error, instead of a warning?  And is it intentional that Verilator applies this also to files that contain address specifications?

The error is reported at line 1913 of `include/verilated.cpp` (added in commit 918df2e6):
```
if (VL_UNLIKELY(m_end != ~0ULL && m_addr <= m_end)) {
        VL_FATAL_MT(m_filename.c_str(), m_linenum, "",
                    "$readmem file ended before specified final address (IEEE 2017 21.4)");
    }
```

I can work around the issue by adding an address specification to the end of my file:
```
...
// return to entry 3
@3
33

// put the address back to the end
@8
```
But this uses an address specification that is outside the specified range.  Verilator accepts this file without even a warning; but the spec seems to say that Verilator should report an error:
> When addressing information is specified both in the system task and in the data file, the addresses in the data file shall be within the address range specified by the system task arguments; otherwise, an error message is issued, and the load operation is terminated.

Another workaround is to use an address and a value, at the end of the file:
```
...
// return to entry 3
@3
33

// put the address back to the end
@7
77
```
In my example, though, I've already specified the contents of that final location, so this is a duplicate write.  (I don't see anything in the spec about warning or erroring if the same location is written multiple times, possibly with different values.  Verilator doesn't report anything.)

wsnyder, 2021-11-17:
Good points.  Perhaps you'd be willing to make a pull request to fix this - that is:

- [ ]  a test
- [ ] simply set a flag on an address being used and then suppress the warning
- [ ] change the error to a warning

quark17, 2021-11-23:
Yes, I will submit a PR for this.  I just need to work on getting the repo to build and run the tests and familiarizing myself with the testing setup.  Sorry for the delay in responding to this and other issues.

closed 2021-12-22 by wsnyder