verilator/verilator #3207: Display of empty string appears as space

quark17, 2021-11-17:
I notice that this statement:
```
$display(">%s<", "");
```
will print `> <` (with a space) in Verilator, but `><` (no space) by Modelsim and Icarus.  The SV 2017 spec is very brief on `%s`, but I assume that printing no space is preferable.

I see that the generated C++ has this:
```
VL_WRITEF("> <\n");
```
so I assume that something in `V3EmitCFunc.cpp` has determined that the string is constant and can be inlined, but I can't see where this is happening.

Here is a self-contained example: [display_empty.tar.gz](https://github.com/verilator/verilator/files/7552568/display_empty.tar.gz)

wsnyder, 2021-11-17:
Looks like this is a bug dating back to pre-system verilog when "" couldn't have zero width.
Will look at it further later.

quark17, 2021-11-17:
FYI, this is boiled down from an example that was written in 2011, with expected output, so I can see that Icarus was not printing the space back then.  That's interesting about the pre-SV Verilog width -- I forgot to look at the older specs.  Is it the case that `""` is represented with `"\0"`?  That is, even though it's required to have width, the first byte is null, so should the `$display` not print anything?  I guess that's a question of how `"\0"` should be displayed.  That suggests testing these statements as well:
```
$display(">%s<", "\0");
$display(">\0<");
```
When I add those to the attached test, I see the additional output:
```
> <
> <
```
With Icarus and Questa I see:
```
><
>
```
So I guess the real issue here is the interpretation of printing `\0`.  Does that have a definitive answer (according to the SV/V specs)?  Verilator's behavior (treating it as a character to be printed and not a string terminal) would not be unreasonable, as long as that's consistent: I see, for example, that if a string is assigned to a signal that's larger, the extra is padded with zeros; so what does printing that signal do?  If I write this:
```
  wire [31:0] r;
  assign r = "ab";
  ...
      $display(">%s<", r);
```
then I get this behavior from Verilator:
```
>  ab<
```
but I also get that behavior from Icarus and Questa!  So they seem to be the ones that are inconsistent?

quark17, 2021-11-17:
I also tried this test:
```
assign r = "a\0b";
```
and saw that Verilator and Icarus printed this:
```
> a b<
```
but that Questa printed this:
```
>   a<
```
So Questa is interpreting the null as the end of the string, but still padding to the length of the string -- which means that `""` should have printed a space, if it was being treated as a non-zero width.

wsnyder, 2021-11-17:
>wire [31:0] r;   assign r = "ab";

That's specified by IEEE to be "   ab".

>assign r = "a\0b";

Verilator's right.  Questa is broken.  Nulls don't terminate as in C.

I think "" should be a string based on SV, as opposed to in Verilog-non-SV where's it's a 8'b0.  I looked at this and a straightforward hack for "" seemed to work, so committed that.  Ideally the code would treat all "...." as a string data type, but that seems to break a lot of code so is something left for later - should be harmless as far as this issue goes, but suggests might be several places where strings are not properly accepted and converted to numeric types, I'll file another bug.

closed 2021-11-17 by wsnyder