verilator/verilator #3208: Missing bits in display of a concat

quark17, 2021-11-17:
I have an example with this:
```
$display("%x", t2);
$display("%x", { t2[15:8], 4'd0, t2[3:0] });
```
and Modelsim and Icarus output this as expected:
```
ff0f
ff0f
```
but Verilator seems to lose the upper bits in the second display:
```
ff0f
0f
```
I see that the following code was generated:
```
VL_WRITEF("%x\n%x\n",16,vlTOPp->display_concat__DOT__t2,
          8,(0xff0fU & (IData)(vlTOPp->display_concat__DOT__t2)));
```
I would guess that the `8` for the second argument is the problem.  Something must have computed the wrong size?

Here is the self-contained example: [display_concat.tar.gz](https://github.com/verilator/verilator/files/7552767/display_concat.tar.gz)

wsnyder, 2021-11-17:
Something is off in V3Width most likely.  Would you be able to take a look?  Otherwise I will look into it.

quark17, 2021-11-23:
I am still trying to respond to your comments on the issues that've opened; I also have a few other examples that I will probably file as new issues (when I can get to that); but first I wanted to submit a PR (at least for the readmem issue), so that I'm contributing and not just blasting you with issues.  I can certainly add this issue to my plate, too, as long as you're OK with the rate which I seem to be clearing things off that plate?  I don't have any idea how this code works, but I'd be interested to look and learn.

wsnyder, 2021-11-23:
In general would love to have you looking into how things work & learn how to fix some of them.  For this one I looked at what was breaking and was easy just to fix.

closed 2021-11-23 by wsnyder