verilator/verilator #3210: Set the L2-name scope from the C++ instantiation

quark17, 2021-11-18:
I have a module `mod` which I instantiate with the instance name `top` in a Verilog testbench module `main`.  When I display the module heirarchy/scope with `%m`, the output I expect from Verilog simulators is `main.top`.  And if I display the hierarchy in a submodule instance `foo`, then I expect `main.top.foo`.  (And similarly for the hierarchy in generated VCD files.)

I am Verilating module `mod` and instantiating it in a C++ testbench, and I'd like for the model to behave as if it has the same instantiated hierarchy as the Verilog testbench.  I notice that the constructor for `VerilatedModule` takes an argument, which seems to be for specifying a scope.  If I omit the argument:
```
Vmod* TOP = new Vmod;
```
then I see output like `TOP.mod.foo` from `%m`.  If I provide an argument, it seems to only be providing the scope of the parent module, because this:
```
Vmod* TOP = new Vmod ("main";
```
leads to output like `main.mod.foo`.

I notice that there is a command-line flag `-l2-name` which can set the instance name.  When I provide `-l2-name top` and provide `"main"` as the constructor argument, then I get the hierarchy that I want: `main.top.foo`.

Is this the only way to get the hierarchy/scope that I want?  Is there a way to do this where the C++ code specifies the name that it wants to use for the instantiation?  The same module can be instantiated in lots of different scopes, which should be allowed to provide different instantiation names.  It seems like Verilated modules hardcode the name of their instance, requiring that name to always be used.  For example, I see the following in the generated code:
```
Vmod__Syms.cpp:    __Vscope_mod__fifo.configure(this, name(), "mod.foo", "fifo", -12, VerilatedScope::SCOPE_OTHER);
...
Vmod__Trace__Slow.cpp:        tracep->declBit(c+26,"mod foo CLK", false,-1);
```

wsnyder, 2021-11-18:
The expectation is the passed in name to the constructor is the name of the "cell" not the top, similar to a SystemC SC_CELL.  So if you have multiple models you'd want to call it with "main.top".

quark17, 2021-11-23:
I apologize that I hadn't responded yet, but I still have questions on this -- can you re-open it? (it doesn't appear that I am able to?)

If passing `"main.top"` to the constructor is what I should do, does that mean that the value for `%m` in display statements is wrong (in that it includes the module name as a level of hierarchy)?  I will attach an example, so we can talk concretely.  I also want to test VCD generation for the example, to see whether the hierarchical names in the VCD include the extra name.  (And I also wanted to look in the test suite, to see what tests there are for `%m`, first, which is also why I hadn't yet responded here yet.  Sorry!)

quark17, 2021-11-25:
Here is an example: [display_scope.tar.gz](https://github.com/verilator/verilator/files/7601128/display_scope.tar.gz)

The Verilog defines a module called `display_scope`.  My intention is that the simulation should instantiate that module with the instance name `top` inside a parent module called `main`.  The module has a register `done` inside it, so if I look at the VCD dump, I expect to see a signal `main.top.done`.  The module contains a display statement with `%m` and I expect that to print `main.top`.

I have followed your advice and passed `"main.top"` as the argument to the constructor.  When I do that, I see this displayed:
```
scope is main.top.display_scope
```
And I see this in the VCD file (which can be turned on by adding `+bscvcd` to the command-line):
```
 $scope module main.top $end
  $var wire  1 # CLK $end
  $var wire  1 $ RST_N $end
  $scope module display_scope $end
   $var wire  1 # CLK $end
   $var wire  1 $ RST_N $end
   $var wire  1 % done $end
   $var wire  1 ' done$D_IN $end
   $var wire  1 & done$EN $end
  $upscope $end
 $upscope $end
```
This is not what I want.  There is a level of hierarchy called `display_scope` which should be called `top`.  And I'm surprised to see signals `CLK` and `RST_N` in the parent, but I assume those are the port connections that would be in the parent, and I'd like that parent to just be named `main`.

I am, however, able to get what I want by passing `"main"` as the argument to the constructor and calling Verilator with the flag `-l2-name top`.  But that seems to go against your explanation of how it should work.

wsnyder, 2021-11-25:
If using the l2 name as you indicate works, that's fine. 

Background for why it is as it is.

1. The reason we need the cell name is imagine your C code instantiates the model as both topA and topB
2. The reason we need the l2 name is in your verilog you can declare symbols at the root, or have multiple modules at the top.

Putting them together in [display_scope.tar.gz](https://github.com/verilator/verilator/files/7603175/display_scope.tar.gz)

```
scope is main.topa.other_scope
scope is main.topa.display_scope
scope is main.topa.$unit
scope is main.topb.other_scope
scope is main.topb.display_scope
scope is main.topb.$unit

```

quark17, 2021-11-25:
Thank you.

I was still very confused by what you wrote.  And I think the explanation for my disconnect may be this (please let me know if I've misunderstood, though):

It seems that there is an implied `module..endmodule` block around all Verilog code provided to Verilator.  And even if I use the `--top-module` flag to insist that there is only one top module, there is still an implied extra layer of module around it.

Or maybe my confusion is this: When the C++ constructor is called in `sim_main.cpp`, I think of that as a module instantiation.  (And the argument is the instance name of the module that I am instantiating.)

Your example is helpful, thank you!  There is a top-level task, and its scope appears as `main.topb.$unit`.  That surprised me, because I would expect a top-level task to not be under any hierarchy.  But if I think of Verilator as implicitly adding a `module..endmodule` block around all of the Verilog code, then that explains the behavior seen in the example.

I'll accept that's how Verilator works.  But I'm not convinced it was necessary.  I think I would have expected that either (a) Verilator would insist on one top module (and the user can write an explicit `module...endmodule` block around all their code, if that's what they want) or else (b) Verilator would generate a separate C++ class for each top module, that could each be instantiated in the C++ code, passing in a name for the instance as the argument.

In any case, would be it possible to direct Verilator to not add the level of hierarchy, when there is only one top module?

I ask because -- while, yes, using `--l2-name` in my way "works" -- it only works for one instantiation.  I have no way of instantiating two copies of module `display_scope` as `topa` and a `topb` without the extra level of hierarchy.  That is, I can't generate this:
```
main.topa.done
main.topb.done
```
A flag to allow the `--top-module` to be the very top would be helpful.

quark17, 2021-11-25:
Separately, I assume that this `toptask` would be another example for #3211.  Since I would expect the `%m` output to contain `toptask` (and I'm not sure where `$unit` came from?).  (If you're curious, Questa prints `display_scope_v_unit.toptask` as the scope.  And Icarus rejects tasks declared outside of modules.)

wsnyder, 2021-11-26:
Is there a reason the extra scope breaks something?  (Versus "looks off")?

Note you can pass just "" to the cell name, and that will result in %m names that should match other simulators.  But then you can't have multiple cells/different cell names which seems you are interested in.

I understand what you want, but this might be complicated to implement as the hierarchical name is fairly important to how the internals resolve signal names.  It'll require some thought.

I think in addition to no MULTITOP modules, if we remove the l2 name, you couldn't have classes, and can't have DPI nor VPI.  That seems fairly restrictive.

> I assume that this toptask would be another example for #3211
Correct.

quark17, 2021-11-26:
> Note you can pass just "" to the cell name, and that will result in %m names that should match other simulators.

If the verilated module is the top module, that's true.  If my top module is
```
module mod();
...
endmodule
```
then, yes, typically in any Verilog simulator I expect the top hierarchy to be `mod` and to see scopes like `mod.foo.bar`.

But in my case I have additional Verilog files that are not being verilated.  I have a file `main.v` containing a module `main` that instantiates a DUT `mod` and twiddles the clock and reset signals:
```
module main();
  wire CLK, RST_N;

  mod top(.CLK(CLK), .RST_N(RST_N));

  ...
endmodule
```
If I simulate this in Verilog simulators, I see scopes like `main.top.foo.bar`.

If I could verilate the module `main`, then, yes, my C++ code could instantiate that with `""` as the argument (or with no argument) and the "l2" name would be `main`, and the scopes would look as expected.

But my understanding is that `main` can not be verilated, and so I have a `main.cpp` file which replaces `main.v` when using Verilator.  When using Verilator with `main.cpp`, I would like the simulation behavior to look like that of Verilog simulators with `main.v`.  And that's where I run into problems.

I have attached an example ([hierarchy.tar.gz](https://github.com/verilator/verilator/files/7610766/hierarchy.tar.gz)) that contains both the `main.v` and the `main.cpp`.  The Makefile builds with Verilator by default, but if you run `make SIM=icarus` then it will build and run with the Icarus simulator.

> Is there a reason the extra scope breaks something? (Versus "looks off")?

It is not the end of the world, but I have two reasons.

I have [a tool](https://github.com/B-Lang-org/bsc) that generates Verilog, with a testsuite of examples and expected behavior.  The CI runs the testsuite with Icarus, but it can be run with other simulators.  The tool generates Verilog for a top module (like `mod`) and then tests are run by adding `main.v` to twiddle the clock and reset.

I'd like to be able to run the testsuite with Verilator.  The reason why I had a sudden spurt of issues to file is because I got Verilator running and discovered differences on a lot of tests.  (And many tests don't verilate -- I need to work through those to see whether they are bugs/limitations of Verilator or if the design needs to be expressed differently; I'll need to learn how designs that create clock gates / clocks can be specified for Verilator.)

Reason one, then, is to make it easier to swap Verilator in as the simulator for the testsuite.  (But I can add special code to the testsuite, to expect slightly different behavior -- it's not the end of the world.)

The tool can also generate its own simulation executable, which is implemented by generating C++ code for the model (rooted at the top module) and linking that into a `main` that twiddles the clock and reset (creating a simulation that behaves similar to the Verilog with `main.v`).  The C++ model could also be instantiated in someone's own C++ project.  And, in fact, the tool can auto-generate a SystemC wrapper for the model, so that it can be used in a SystemC project.  (The model can be instantiated multiple times, with different instance names.  Two different models, that were generated by separate invocations of the tool, can also be instantiated in the same C++ or SystemC project.)

Reason two, then, is that I was curious whether Verilator could replace this C++ backend.  (Or become a third backend.)

> I think in addition to no MULTITOP modules, if we remove the l2 name, you couldn't have classes, and can't have DPI nor VPI.

I don't see any fundamental reason why you couldn't do any of these (multi-top, classes, DPI), but I'd like to understand.

Here are the situations I'm used to:

(a) The Verilog files are the entire universe.  Verilog simulators like Icarus and Questa, will find any modules without instantiations, consider them to be the roots, and will create a simulation that has a top-level instance for each, using the module name as the hierarchical name.  (Users can also specify the roots, too, I guess.)

(b) The Verilog files are not the entire universe; there is an actual root somewhere else that will instantiate a module (or modules) defined in the Verilog files.  For the tool described above, the user specifies a module in the Verilog files, and the tool generates code for hierarchy rooted at that module, which can be instantiated inside the actual root.

I don't see why the same code generation for a root module can't support both scenarios: for A, the object is instantiated with its module name as its hierarchy (e.g. `main`); for B, the object is instantiated with its hierarchy within the larger hierarchy (e.g. `main.top`).

Verilator seems to be supporting a third scenario, possibly a hyrbid of the two, where any uninstantiated modules (and any tasks or DPI not inside modules) are instantiated by their module name as roots (like scenario A) but they are then put inside a hierarchy which can be instantiated by C++ code (like scenario B but with an extra level of hierarchy added).

A tool that generates C++ code for hierarchy rooted at a module, that can be instantiated with an instance name (or top-level name), without any extra hierarchy, should in theory be able to support DPI and multitop (because the tool mentioned above does -- granted, there are limitations on the signals at the top level, for example you can't have  [... cut ...] etween the modules).

(For our tool, if there are multiple roots that you want to create models for, to instantiate inside a C++ or SystemC project, then you make two independent invocations of the tool, specifying module X the first time and then module Y the second time.  These could generate independent objects; however our current implementation has a common kernel provided as a library, that is instantiated once, and the model constructors are passed a pointer and connect to it via an API.)

quark17, 2021-11-27:
> I don't see why the same code generation for a root module can't support both scenarios

By which I mean that, places like this in the code:
```
tracep->declBit(c+26,"mod foo CLK", false,-1);
```
instead of hardcoding `mod` as the first level of the hierarchy, could construct the string with `"%s foo CLK"` and fill in the first level with the string that was passed to the constructor -- either the module name `mod` (if it's scenario A) or `main.top` (if it's scenario B).

wsnyder, 2021-11-27:
I understand what you're after.  I don't think this is easy to support but let me think about it.