verilator/verilator #3213: How do I configure fst traces for maximum performance?

oharboe, 2021-11-19:
I've been outputting .vcd files and converting them to .fst using vcd2fst in gtkwave.

However, I thought it would be a good idea to output .fst files directly from Verilator.

The problem is that vcd2fst is substantially faster than generating .fst files directly from Verilator. This is with Verilator from Ubuntu 21.10, so Verilator 4.038.

In my case, Verilator simulation and output .vcd file 25s, vcd2fst 25s, launching gtkwave on .vcd file 44s.

The fastest way to open gtkwave is to generate .vcd file, then launch gtkwave with -o that runs vcd2fst first.

oharboe, 2021-11-19:
On my laptop, generating .fst file during Verilator simulation takes 15 minutes. On my workstation, generating .fst file during Verilator simulation takes 4 minutes.

I tried adding --trace-threads=6, but it had no impact on my workstation.

My workstation has 24 cores(48 threads), so 6 trace threads and 6 simulation threads should not overload it.

gezalore, 2021-11-19:
The manual explains why --trace-theads should be 1 or 2 for FST (2 might not be faster on some designs): https://verilator.org/guide/latest/exe_verilator.html?highlight=trace%20threads#cmdoption-trace-threads

As of the speed difference between platforms: Any chance your laptop is ARM based? The tracing code uses x86 intrinsics for speeding up some parts, which is not currently ported to other architectures.

oharboe, 2021-11-19:
My laptop is an x86 i7 6 cores and 12 threads and my workstation is a threadripper with 24 cores and 48 threads.

I tried with --trace-threads 2, no difference.

Verilator simulation without writing a .vcd or .fst file takes 4 seconds.

gezalore, 2021-11-19:
> I tried with --trace-threads 2, no difference.

Could you be a bit  more precise? no difference compared to what?

Are you running into https://github.com/verilator/verilator/issues/3168 ? (flushing the trace file too often)

oharboe, 2021-11-19:
I'm using Verilator 4.038, so a lot older than the fix in #3168.

I need to chase down things more, evidently...

gezalore, 2021-11-19:
There is no fix in #3168. If you have a `tracep->flush()` in the inner loop of your C++, tracing will be slow. The fix is don't do that.

oharboe, 2021-11-19:
I'm closing this one:

1. I have made some mistakes I need to correct
2. I need to test with the latest Verilator.

If the problem persists, I will create a fresh issue.

closed 2021-11-19 by oharboe

oharboe, 2021-11-19:
@gezalore  I have more than one thing to fix, but also the tracep->flush() fix you mention above. Thanks!