verilator/verilator #3215: Internal error on string parameter in latest master

adrienlemasle, 2021-11-24:
The following code:
```
module test;
   parameter A = "";

   initial begin
      if(A == "TEST") begin
         $display("Test");
      end
   end
endmodule
```

leads to the following error on latest master:
```
%Error: Internal Error: test.sv:6:12: ../V3Number.cpp:2433: Number operation called with non-string argument: '64'h54455354"
                                    : ... In instance test
    6 |       if(A == "TEST") begin
      |            ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

This is working properly on latest stable version (4.214).

wsnyder, 2021-11-24:
This is almost certainly a side effect of fixing #3207, I'll fix it.

Note this code probably isn't doing quite what you expect, A defaults to the "bit [7:0]" type, which doesn't match ""'s type of string of width zero, hence the error.  Assuming you are using SystemVerilog you're better off writing "parameter string A = ""; '.

adrienlemasle, 2021-11-24:
This is not in my code actually. There is plenty of this in Xilinx XPM
libraries and they have been working fine with previous stable versions of
Verilator.

wsnyder, 2021-11-24:
Yes, agree the code is legal, just not sure if people realize per IEEE it isn't doing a string comparison.

adrienlemasle, 2021-11-24:
Interesting. I'll make sure to use a string if I ever use this syntax in my
code. Thanks for the great work on Verilator btw :-)

wsnyder, 2021-11-25:
Should be good, please try it in case something else is off.

closed 2021-11-25 by wsnyder

adrienlemasle, 2021-11-25:
Yes looks good now thanks!

goekce, 2022-12-26:
> ```sv
> module test;
> parameter A = "";
> ...
> ```

> ... A defaults to the "bit [7:0]" type, which doesn't match ""'s type of string of width zero, hence the error. Assuming you are using SystemVerilog you're better off writing "parameter string A = ""; '.

I assumed Systemverilog would decide the type based on the initialized value so I researched about this in LRM 2017.

6.20.2:
> A parameter declaration with no type or range specification shall default to the type and range of the final value assigned to the parameter, after any value overrides have been applied. If the expression is real, the parameter is real. If the expression is integral, the parameter is a logic vector of the same size with range `[size-1:0]`.

6.16:

> A string variable does not represent a string in the same way as a string literal (see 5.9). String literals behave like packed arrays of a width that is a multiple of 8 bits. A string literal assigned to a packed array of an integral variable of a different size is either truncated to the size of the variable or padded with zeros to the left as necessary. When using the string data type instead of an integral variable, strings can be of arbitrary length and no truncation occurs. String literals are implicitly converted to the string type when assigned to a string type or used in an expression involving string type operands.

> A string literal can be assigned to a variable of a string or an integral data type. ...

There is a problem: The datatype of a type-unspecified parameter depends on the type of the initial value. At the same time string literals can both be assigned to integral types and `string`s. The LRM is not specific enough here IMHO.

I can understand that verilator opts for an integral type as default, but providing the type explicitly in following example should deduce a `string`.

```sv
module m;                                                                                                                                                                                                                                    
parameter A = string'("aaaa");                                                                                                                                                                                                               
initial begin                                                                                                                                                                                                                                
        $display($typename(A));                                                                                                                                                                                                              
        if(A == "TEST")                                                                                                                                                                                                                      
                $display("Test");                                                                                                                                                                                                            
end                                                                                                                                                                                                                                          
endmodule
```
outputs
```
$ verilator --binary -j $(nproc) -GA=\"TEST\" m.sv
$ obj_dir/Vm
logic[31:0]
Test
```
What do you think @wsnyder ?

wsnyder, 2022-12-26:
Ibelieve Verilator is correct, and based on previous experiments matches most other simulators.  "" defaults to integral unless it is specifically assigned to a string. IEEE did this (AFAIK) to maintain backwards compatibility with V-2001.