verilator/verilator #3219: Help: Fix UNOPTFLAT in a case statement?

zangman, 2021-11-26:
How may we help - what is your question?
I am designing a CPU which needs the memory to be byte addressable but has 4-byte wide addressing (risc-v). To ensure it gets synthesized to BRAM, I've created 4 separate 1 byte-wide rams which are managed by a memory controller.

However, if I want to access unaligned memory i.e. not multiples of 4, I need to permute the values I read from the memory modules accordingly.

This has the side effect of the read data depending directly on the address in a combinational block thus causing `UNOPTFLAT`.

I've encountered this issue many times and most times I could split my code to be in a different `always` block and that fixed it. However, here I have a `case` statement which I can't split out.

Here is the code which is causing the problem:

```verilog
assign offset = a[1:0];
assign a_by_4 = a >> 2;
assign a_plus_1 = a_by_4 + 1;

always_comb begin
  case(offset)
    2'd0: rd = {rd3, rd2, rd1, rd0};
    2'd1: rd = {rd0, rd3, rd2, rd1};
    2'd2: rd = {rd1, rd0, rd3, rd2};
    2'd3: rd = {rd2, rd1, rd0, rd3};
  endcase
end
```
The error message is quite clear and I understand the problem:

```bash
verilator -Wall -cc -build --trace rtl/top.sv -Irtl
%Warning-UNOPTFLAT: rtl/top.sv:42:15: Signal unoptimizable: Feedback to clock or circular logic: 'top.mem_adr'
   42 | logic [31:0]  mem_adr;
      |               ^~~~~~~
                    ... For warning description see https://verilator.org/warn/UNOPTFLAT?v=4.215
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
                    rtl/top.sv:42:15:      Example path: top.mem_adr
                    rtl/mem_controller.sv:78:1:      Example path: ALWAYS
                    rtl/top.sv:30:15:      Example path: top.mem_rd
                    rtl/mux2.sv:8:12:      Example path: ASSIGNW
                    rtl/top.sv:29:15:      Example path: top.inst
                    rtl/control.sv:128:1:      Example path: ALWAYS
                    rtl/top.sv:18:15:      Example path: top.c_adrsrc
                    rtl/mux2.sv:8:12:      Example path: ASSIGNW
                    rtl/top.sv:42:15:      Example path: top.mem_adr
%Error: Exiting due to 1 warning(s)
make: *** [Makefile:31: top] Error 1

```

Unfortunately, I don't know how to fix it. Steps that I tried:

 1. Follow along and split various signals into separate `always` blocks. Keep doing this and I end up at another place where I have a similar case like the above.
 2. Tried the comment `split_var` on all the signals in the generated by `--report-unoptflat`. That didn't work either.
 3. As a last option, tried `lint_off` for unoptflat. That worked fine for aligned memory access but threw `DIDNOTCONVERGE` error for unaligned access.

In the code above, if I remove the dependency on `offset`, the error goes away i.e. :

```verilog
always_comb begin
  rd = {rd3, rd2, rd1, rd0};
end
```

Any suggestions on how I can get fix this? I can't think of any way to remove the dependency of `rd` on `offset`. If there isn't a way, I might need to either try a completely different approach for memory or drop unaligned access completely.

Thanks in advance!

wsnyder, 2021-11-26:
I see nothing wrong with the case statement as far as causing an UNOPTFLAT.  I suspect what is happening is when there is no case statement Verilator optimizes the statements into other logic and is able to resolve away the warning.

The warning is not only saying there is a path from mem_adr to mem_rd (obviously needed), but also there's a path from mem_rd back to the mem_adr.  I'd think that's the one that should be broken as seems unusual and unlikely to make timing to have read data affect the address without a flop in the middle.

If you can make a test case with everything along the path I can run, I can probably make a better recommendation.

zangman, 2021-11-26:
>  I suspect what is happening is when there is no case statement Verilator optimizes the statements into other logic and is able to resolve away the warning.

I see. So the issue arising with the case statement is a red herring and the problem could be somewhere else? If that's the case, let me have a go at optimizing the rest of it once more and see if there is anything I can do.

> there's a path from mem_rd back to the mem_adr. I'd think that's the one that should be broken as seems unusual and unlikely to make timing to have read data affect the address without a flop in the middle.

So the memory contains both instructions and data which are made available on `rd` at different addresses. The control module has a bunch of combinational logic including one which goes to a mux which decides where the next address should be assigned from (program counter or from the ALU). In theory it should not affect anything because `rd` is synchronous and only gets updated on clock edge. But perhaps that's not what's happening since I am getting a `DIDNOTCONVERGE` error.

> If you can make a test case with everything along the path I can run, I can probably make a better recommendation.

Thank you! I will try and see if I can make a test case. The difficulty is that when I resolve this particular one by splitting the signals appropriately, there are new ones that show up.

wsnyder, 2021-11-26:
>So the issue arising with the case statement is a red herring and the problem could be somewhere else?

That's my leading theory.

zangman, 2021-11-27:
I think I may have solved this. It's a solution specific to my implementation, so I don't know if there is a larger learning here. Or even if this is a valid solution to the problem.

I realized that since the memory is synchronous, `rd` is never valid before the cycle after `a` is asserted. So there's no harm in breaking the combinational loop above with a synchronous one. So I added `offset_d` which is synchronous and that made the error go away.

```verilog
always_ff @(posedge clk) offset_d <= offset;

always_comb begin
  case(offset_d)
    2'd0: rd = {rd3, rd2, rd1, rd0};
    2'd1: rd = {rd0, rd3, rd2, rd1};
    2'd2: rd = {rd1, rd0, rd3, rd2};
    2'd3: rd = {rd2, rd1, rd0, rd3};
  endcase
end
```

My tests against an emulator pass and I have unaligned access without any added clock cycles, so it's definitely a win. Only question is if this is a verilator specific hack. I am not too familiar with icarus or modelsim so I don't know if this would be a problem there as well. Or if it is generally good practice to avoid circular combination loops in hardware.

wsnyder, 2021-11-27:
Makes sense.  Yes, generally a combo loop is to be avoided, though Verilator is much pickier than most tools.

closed 2021-11-27 by wsnyder