verilator/verilator #3223: make -j126 failed in verilator@4.214

Tom-python0121, 2021-11-29:
hello,I meet a problem:make -j126 failed in verilator@4.214

### Steps to reproduce the issue

```console
……
/usr/include/c++/7/bits/stl_pair.h:350:21: note: candidate: template<class _U1, class _U2, typename std::enable_if<(_MoveConstructiblePair<_U1, _U2>() && (! _ImplicitlyMoveConvertiblePair<_U1, _U2>())), bool>::type <anonymous> > constexpr std::pair<_T1, _T2>::pair(_U1&&, _U2&&)
  explicit constexpr pair(_U1&& __x, _U2&& __y)
                     ^~~~
/usr/include/c++/7/bits/stl_pair.h:350:21: note:   template argument deduction/substitution failed:
/usr/include/c++/7/bits/stl_pair.h:349:38: error: no type named 'type' in 'struct std::enable_if<false, bool>'
                          bool>::type=false>
                                      ^~~~~
/usr/include/c++/7/bits/stl_pair.h:349:38: note: invalid template non-type parameter
/usr/include/c++/7/bits/stl_pair.h:341:12: note: candidate: template<class _U1, class _U2, typename std::enable_if<(_MoveConstructiblePair<_U1, _U2>() && _ImplicitlyMoveConvertiblePair<_U1, _U2>()), bool>::type <anonymous> > constexpr std::pair<_T1, _T2>::pair(_U1&&, _U2&&)
  constexpr pair(_U1&& __x, _U2&& __y)
            ^~~~
/usr/include/c++/7/bits/stl_pair.h:341:12: note:   template argument deduction/substitution failed:
/usr/include/c++/7/bits/stl_pair.h:340:38: error: no type named 'type' in 'struct std::enable_if<false, bool>'
                          bool>::type=true>
                                      ^~~~
/usr/include/c++/7/bits/stl_pair.h:340:38: note: invalid template non-type parameter
/usr/include/c++/7/bits/stl_pair.h:332:17: note: candidate: template<class _U2, typename std::enable_if<_CopyMovePair<false, AstNodeModule*, _U2>(), bool>::type <anonymous> > std::pair<_T1, _T2>::pair(const _T1&, _U2&&)
        explicit pair(const _T1& __x, _U2&& __y)
                 ^~~~
/usr/include/c++/7/bits/stl_pair.h:332:17: note:   template argument deduction/substitution failed:
/usr/include/c++/7/bits/stl_pair.h:331:38: error: no type named 'type' in 'struct std::enable_if<false, bool>'
                          bool>::type=false>
                                      ^~~~~
/usr/include/c++/7/bits/stl_pair.h:331:38: note: invalid template non-type parameter
/usr/include/c++/7/bits/stl_pair.h:325:18: note: candidate: template<class _U2, typename std::enable_if<_CopyMovePair<true, AstNodeModule*, _U2>(), bool>::type <anonymous> > constexpr std::pair<_T1, _T2>::pair(const _T1&, _U2&&)
        constexpr pair(const _T1& __x, _U2&& __y)
                  ^~~~
/usr/include/c++/7/bits/stl_pair.h:325:18: note:   template argument deduction/substitution failed:
/usr/include/c++/7/bits/stl_pair.h:324:38: error: no type named 'type' in 'struct std::enable_if<false, bool>'
                          bool>::type=true>
                                      ^~~~
/usr/include/c++/7/bits/stl_pair.h:324:38: note: invalid template non-type parameter
/usr/include/c++/7/bits/stl_pair.h:318:27: note: candidate: template<class _U1, typename std::enable_if<_MoveCopyPair<false, _U1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(), bool>::type <anonymous> > constexpr std::pair<_T1, _T2>::pair(_U1&&, const _T2&)
        explicit constexpr pair(_U1&& __x, const _T2& __y)
……
/usr/include/c++/7/bits/stl_pair.h:229:26: note:   template argument deduction/substitution failed:
In file included from /usr/include/aarch64-linux-gnu/c++/7/bits/c++allocator.h:33:0,
                 from /usr/include/c++/7/bits/allocator.h:46,
                 from /usr/include/c++/7/string:41,
                 from /usr/include/c++/7/bits/locale_classes.h:40,
                 from /usr/include/c++/7/bits/ios_base.h:41,
                 from /usr/include/c++/7/iomanip:40,
                 from ../V3String.h:25,
                 from ../V3Error.h:24,
                 from ../V3Global.h:29,
                 from ../V3LinkDot.cpp:67:
/usr/include/c++/7/ext/new_allocator.h:136:4: note:   candidate expects 0 arguments, 2 provided
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../Makefile_obj:301: recipe for target 'V3Global.o' failed
make[2]: *** [V3Global.o] Error 1
../Makefile_obj:301: recipe for target 'V3EmitCFunc.o' failed
make[2]: *** [V3EmitCFunc.o] Error 1
../Makefile_obj:301: recipe for target 'V3Cdc.o' failed
make[2]: *** [V3Cdc.o] Error 1
../Makefile_obj:301: recipe for target 'V3EmitCInlines.o' failed
make[2]: *** [V3EmitCInlines.o] Error 1
../Makefile_obj:301: recipe for target 'V3AstNodes.o' failed
make[2]: *** [V3AstNodes.o] Error 1
make[2]: Leaving directory '/home/stage/root/spack-stage-verilator-4.214-75es6zkmdnkfg6wfmx4hpzbm74d2o4i7/spack-src/src/obj_opt'
Makefile:59: recipe for target '../bin/verilator_bin' failed
make[1]: *** [../bin/verilator_bin] Error 2
../Makefile_obj:301: recipe for target 'V3AstNodes.o' failed
make[2]: *** [V3AstNodes.o] Error 1
../Makefile_obj:301: recipe for target 'V3EmitCSyms.o' failed
make[2]: *** [V3EmitCSyms.o] Error 1
../Makefile_obj:301: recipe for target 'V3LinkDot.o' failed
make[2]: *** [V3LinkDot.o] Error 1
make[2]: Leaving directory '/home/stage/root/spack-stage-verilator-4.214-75es6zkmdnkfg6wfmx4hpzbm74d2o4i7/spack-src/src/obj_dbg'
Makefile:65: recipe for target '../bin/verilator_bin_dbg' failed
make[1]: *** [../bin/verilator_bin_dbg] Error 2
make[1]: Leaving directory '/home/stage/root/spack-stage-verilator-4.214-75es6zkmdnkfg6wfmx4hpzbm74d2o4i7/spack-src/src'
Makefile:139: recipe for target 'verilator_exe' failed
make: *** [verilator_exe] Error 2
```
What can I do to get it to compile through

wsnyder, 2021-11-29:
Please show the first/top of the errors rather than the bottom only.

What OS distribution and 'gcc --version' are you using?

Tom-python0121, 2021-11-30:
@wsnyder As you can see the first errors:

```console
ENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Inline.cpp -o V3Inline.o
/home/all_spack_env/spack/lib/spack/env/gcc/g++   -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Inst.cpp -o V3Inst.o
/home/all_spack_env/spack/lib/spack/env/gcc/g++   -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3InstrCount.cpp -o V3InstrCount.o
/home/all_spack_env/spack/lib/spack/env/gcc/g++   -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Inst.cpp -o V3Inst.o
/home/all_spack_env/spack/lib/spack/env/gcc/g++   -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3InstrCount.cpp -o V3InstrCount.o
In file included from ../V3ActiveTop.cpp:29:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3Active.cpp:32:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3AssertPre.cpp:23:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3Begin.cpp:31:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3Cast.cpp:43:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3CUse.cpp:28:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3Depth.cpp:29:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3Dead.cpp:39:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3Active.cpp:32:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
In file included from ../V3EmitCInlines.cpp:20:
../V3Global.h:23:3: error: #error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
 # error "Something failed during ./configure as config_build.h is incomplete. Perhaps you used autoreconf, don't."
   ^~~~~
……
[root@localhost spack-src]# gcc --version
gcc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-4)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
How can I fix it?

wsnyder, 2021-11-30:
Again, what's your OS distribution?

As it says, something failed during configure.  What did you get from running "./configure"?

Tom-python0121, 2021-11-30:
```console
[root@localhost spack-src]# ./configure
configuring for Verilator 4.214 2021-10-17
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to use -m32... no
checking whether to build for coverage collection... no
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... /home/all_spack_env/spack/lib/spack/env/gcc/gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/gcc accepts -g... yes
checking for /home/all_spack_env/spack/lib/spack/env/gcc/gcc option to accept ISO C89... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -g... yes
checking for a BSD-compatible install... /usr/bin/install -c
compiler is /home/all_spack_env/spack/lib/spack/env/gcc/g++ --version = g++ (GCC) 8.5.0 20210514 (Red Hat 8.5.0-4)
checking that C++ compiler can compile simple program... yes
checking for ar... ar
checking for perl... /home/all_spack_env/spack/opt/spack/linux-centos8-aarch64/gcc-8.5.0/perl-5.32.1-of4eesom2rgrehio7qlkx2wne6tzz5gm/bin/perl
checking for python3... /usr/bin/python3
checking for flex... /home/all_spack_env/spack/opt/spack/linux-centos8-aarch64/gcc-8.5.0/flex-2.6.4-fxjlrnwqb3vl5c4f3bbn3rjiad5zg3l4/bin/flex
/home/all_spack_env/spack/opt/spack/linux-centos8-aarch64/gcc-8.5.0/flex-2.6.4-fxjlrnwqb3vl5c4f3bbn3rjiad5zg3l4/bin/flex --version = flex 2.6.4
checking for bison... /home/all_spack_env/spack/opt/spack/linux-centos8-aarch64/gcc-8.5.0/bison-3.7.4-mtty5s43wxmsi7fxyra7j3hjn2bgyvsa/bin/bison
/home/all_spack_env/spack/opt/spack/linux-centos8-aarch64/gcc-8.5.0/bison-3.7.4-mtty5s43wxmsi7fxyra7j3hjn2bgyvsa/bin/bison --version = bison (GNU Bison) 3.7.4
checking for ccache... no
checking how to run the C++ preprocessor... /home/all_spack_env/spack/lib/spack/env/gcc/g++ -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for size_t... yes
checking for size_t... (cached) yes
checking for inline... inline
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -pg... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -std=gnu++14... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wextra... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wfloat-conversion... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wlogical-op... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wthread-safety... no
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Qunused-arguments... no
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -faligned-new... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wno-unused-parameter... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wno-shadow... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wno-char-subscripts... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wno-null-conversion... no
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wno-parentheses-equality... no
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Wno-unused... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -Og... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -ggdb... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -gz... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ linker accepts -gz... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -faligned-new... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -fbracket-depth=4096... no
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -fcf-protection=none... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ accepts -mno-cet... no
[... 14 lines cut ...]
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ linker accepts -static-libgcc... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ linker accepts -static-libstdc++... no
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ linker accepts -Xlinker -gc-sections... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ linker accepts -lpthread... yes
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ linker accepts -lbcrypt... no
checking whether /home/all_spack_env/spack/lib/spack/env/gcc/g++ linker accepts -lpsapi... no
[... cut ...]
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/Makefile_obj
config.status: creating include/verilated.mk
config.status: creating include/verilated_config.h
config.status: creating verilator.pc
config.status: creating verilator-config.cmake
config.status: creating verilator-config-version.cmake
config.status: creating src/config_build.h
config.status: src/config_build.h is unchanged

Now type 'make' (or sometimes 'gmake') to build Verilator.

```

wsnyder, 2021-11-30:
I have no idea how spack works, so you might likely need to debug this yourself.

>config.status: src/config_build.h is unchanged

That seems suspicious, did you really configure it from a new and clean area?

What's the contents of src/config_build.h?

Tom-python0121, 2021-11-30:
@wsnyder  src/config_build.h
```console
/* src/config_build.h.  Generated from config_build.h.in by configure.  */
/* src/config_build.h.in.  Generated from configure.ac by autoheader.  */

/* Define to 1 if you have the <inttypes.h> header file. */
#define HAVE_INTTYPES_H 1

/* Define to 1 if you have the <memory.h> header file. */
#define HAVE_MEMORY_H 1

/* Defined if struct stat has st_mtim.tv_nsec */
#define HAVE_STAT_NSEC 1

/* Define to 1 if you have the <stdint.h> header file. */
#define HAVE_STDINT_H 1

/* Define to 1 if you have the <stdlib.h> header file. */
#define HAVE_STDLIB_H 1

/* Define to 1 if you have the <strings.h> header file. */
#define HAVE_STRINGS_H 1

/* Define to 1 if you have the <string.h> header file. */
#define HAVE_STRING_H 1

/* Defined if have SystemC library */
/* #undef HAVE_SYSTEMC */

/* Define to 1 if you have the <sys/stat.h> header file. */
#define HAVE_SYS_STAT_H 1

/* Define to 1 if you have the <sys/types.h> header file. */
#define HAVE_SYS_TYPES_H 1

/* Define to 1 if you have the <unistd.h> header file. */
#define HAVE_UNISTD_H 1

/* Define to the address where bug reports for this package should be sent. */
#define PACKAGE_BUGREPORT "https://verilator.org"

/* Define to the full name of this package. */
#define PACKAGE_NAME "Verilator"

/* Define to the full name and version of this package. */
#define PACKAGE_STRING "Verilator 4.214 2021-10-17"

/* Define to the one symbol short name of this package. */
#define PACKAGE_TARNAME "verilator"

/* Define to the home page for this package. */
#define PACKAGE_URL "https://verilator.org"

/* Define to the version of this package. */
#define PACKAGE_VERSION "4.214 2021-10-17"

/* Package version as a number */
#define PACKAGE_VERSION_NUMBER_STRING "4.214"

/* Define to 1 if you have the ANSI C header files. */
#define STDC_HEADERS 1

/* Define to `__inline__' or `__inline' if that's what the C compiler
   calls it, or to nothing if 'inline' is not supported under any name.  */
#ifndef __cplusplus
/* #undef inline */
#endif

/* Define to `unsigned int' if <sys/types.h> does not define. */
/* #undef size_t */

```

Tom-python0121, 2021-11-30:
@wsnyder  The same error is reported when verifier@4.214 is compiled on CentOS8_x86_64.

wsnyder, 2021-11-30:
That config_build.h wasn't generated from config_build.h.in.  Either your configure/autoconf is broken, or something you are doing is breaking configure/autoconf.  You'll need to debug this yourself as I have no experience in those being broken.

Alternatively, you can try making a complete new area and following the steps documented to install, without using spack etc, as that seems to be what you are doing that is quite different.

Tom-python0121, 2021-12-01:
@wsnyder you are right,I have compiler it,I will close it

closed 2021-12-01 by Tom-python0121