verilator/verilator #3224: fscanf into structures generates invalid C++ code

adrienlemasle, 2021-12-03:
Verilog code:
```
module test;

   integer file_handle;

   typedef struct packed {
      logic [10:0] a;
      logic [2:0]  b;
   } test_struct_t;

   test_struct_t test;

   initial begin
      integer scan;
      file_handle = $fopen("test.txt", "r");
      scan = $fscanf(file_handle,
                     "%h %h",
                     test.a,
                     test.b);
      $finish;
   end

endmodule
```

C++ code:
```
#include <verilated.h>          // Defines common routines
#include <signal.h>
#include "Vtest.h"               // From Verilating "top.v"

Vtest*top;                      // Instantiation of model

vluint64_t main_time = 0;       // Current simulation time
// This is a 64-bit integer to reduce wrap over issues and
// allow modulus.  This is in units of the timeprecision
// used in Verilog (or from --timescale-override)

double sc_time_stamp() {        // Called by $time in Verilog
    return main_time;           // converts to double, to match
                                // what SystemC does
}

void intHandler(int dummy) {
	printf("Closing...\n");
	exit(0);
}

int main(int argc, char** argv) {
	signal(SIGINT, intHandler);
    Verilated::commandArgs(argc, argv);   // Remember args

    top = new Vtest;             // Create model

    while (!Verilated::gotFinish()) {
        top->eval();            // Evaluate model
        main_time++;            // Time passes...
    }

    //top->final();               // Done simulating
    //    // (Though this example doesn't get here)
    delete top;
}
```

The following command gives:
```
verilator --cc --timescale-override 1ps/1ps test.sv --top-module test --exe --build main.cpp
make: Entering directory 'REDACTED
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o main.o ../main.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtest.cpp Vtest___024root__DepSet_hc07518e4__0.cpp Vtest___024root__Slow.cpp Vtest___024root__DepSet_hc07518e4__0__Slow.cpp Vtest__Syms.cpp > Vtest__ALL.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o Vtest__ALL.o Vtest__ALL.cpp
In file included from Vtest__ALL.cpp:6:
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp: In function ‘void Vtest___024root___initial__TOP__1(Vtest___024root*)’:
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp:23:63: error: lvalue required as unary ‘&’ operand
   23 |                                                         >> 3U)),
      |                                                               ^
make: *** [/usr/local/share/verilator/include/verilated.mk:253: Vtest__ALL.o] Error 1
make: Leaving directory 'REDACTED'
%Error: make -C obj_dir -f Vtest.mk exited with 2
%Error: Command Failed /usr/local/bin/verilator_bin --cc --timescale-override 1ps/1ps test.sv --top-module test --exe --build main.cpp
```

This is for current master.

wsnyder, 2021-12-04:
Given how the code works that makes sense, any packed array reference can't be used where a system call will load a value.  It'll need to add some new VL_ runtime operator, or use a temporary variable on any lvalue in a scanf.

Might be a little complicated as a project, but if you'd be willing to try making a patch I can describe the ideas.

adrienlemasle, 2021-12-06:
Let me see cause this bug is quite easy to work around in testbench code by simply defining intermediate signals as you said. I have another issue related to associative arrays that I'm trying to make a simple testcase for. This might be more interesting.

adrienlemasle, 2021-12-12:
@wsnyder I'll look into this next. Could you please describe the ideas for the fix?

wsnyder, 2021-12-12:
I think we need V3Expand to not convert AstSel to a AstShift when there's a VarRef that's an LValue underneath.
Then the select should get through to C++ code, but we then need some code to handle the insert operation.  I think we want Verilator to put out code to call something like this (the class would go in verilated_types.h)

```
#include <iostream>
#include "verilated.h"

template <typename T_Field, typename T_Out, int T_Lsb, int T_FieldBits> class VlBitSelInsert {
private:
    T_Out& m_outr;  // Reference to output value to be set
    T_Field m_field;  // Field value to be updated

public:
    VlBitSelInsert(T_Out& ref)
        : m_outr{ref} {
        m_field = ref >> T_Lsb;  // Unclean, will get cleaned when re-import
    }
    ~VlBitSelInsert() {
        const QData insmask = (VL_MASK_Q(T_FieldBits)) << T_Lsb;
        m_outr = (m_outr & ~insmask) | ((m_field << T_Lsb) & insmask);
    }
    T_Field* refp() { return &m_field; }
};

// Would also need a wide flavor that can call e.g. _vlinsert_WW

void HACK_VL_SCANF(SData* vp) { *vp = 0x123; }
void HACK_VL_SCANF_UNCHANGED(SData* vp) { }

//======================================================================

int main(int argc, char* argv[]) {
    SData val = 0;
    HACK_VL_SCANF(&val);
    std::cout << "Val = 0x" << std::hex << val << std::endl;

    QData wider = 0xffffffffffffffffULL;
    std::cout << "Wider = 0x" << std::hex << wider << std::endl;
    HACK_VL_SCANF_UNCHANGED(VlBitSelInsert<uint16_t, uint64_t, 4, 12>{wider}.refp());
    std::cout << "Wider = 0x" << std::hex << wider << std::endl;
    HACK_VL_SCANF(VlBitSelInsert<uint16_t, uint64_t, 4, 12>{wider}.refp());
    std::cout << "Wider = 0x" << std::hex << wider << std::endl;

    return 0;
}
```

Note this only shows the integer version, we'd need a template specialization (or something named differently) for wide data (over 64 bits) too.

adrienlemasle, 2026-10-22:
@wsnyder I think this should be reopened. With latest master:

```
module test;
   integer file_handle;

   typedef struct packed {
      logic [10:0] a;
      logic [2:0]  b;
   } test_struct_t;

   test_struct_t test;
   test_struct_t expected;

   initial begin
      integer scan;
      file_handle = $fopen("test.txt", "r");
      scan = $fscanf(file_handle,
                       "%h",
                       test.a);

      if(test != expected) begin
         $display("%h", expected);
      end
      $finish;
   end

endmodule
```
Verilator output:
```
verilator --cc --timescale-override 1ps/1ps test.sv --top-module test --binary
make: Entering directory '/home/adrien/verilator_test/obj_dir'
ccache /usr/local/bin/g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVERILATOR=1 -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -DVM_TRACE_SAIF=0 -DVM_VPI=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-int-in-bool-context -Wno-shadow -Wno-sign-compare -Wno-subobject-linkage -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable  -DVM_PREFIX_INCLUDE="<Vtest.h>"   -DVL_TIME_CONTEXT   -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
ccache /usr/local/bin/g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVERILATOR=1 -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -DVM_TRACE_SAIF=0 -DVM_VPI=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-int-in-bool-context -Wno-shadow -Wno-sign-compare -Wno-subobject-linkage -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable  -DVM_PREFIX_INCLUDE="<Vtest.h>"   -DVL_TIME_CONTEXT   -c -o verilated_threads.o /usr/local/share/verilator/include/verilated_threads.cpp
python3 /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtest.cpp Vtest___024root__0.cpp Vtest__main.cpp Vtest___024root__Slow.cpp Vtest___024root__0__Slow.cpp Vtest__Syms__Slow.cpp > Vtest__ALL.cpp
ccache /usr/local/bin/g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVERILATOR=1 -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -DVM_TRACE_SAIF=0 -DVM_VPI=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-int-in-bool-context -Wno-shadow -Wno-sign-compare -Wno-subobject-linkage -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable  -DVM_PREFIX_INCLUDE="<Vtest.h>"   -DVL_TIME_CONTEXT   -c -o Vtest__ALL.o Vtest__ALL.cpp
In file included from Vtest__ALL.cpp:7:
Vtest___024root__0__Slow.cpp: In function ‘void Vtest___024root___eval_initial(Vtest___024root*)’:
Vtest___024root__0__Slow.cpp:30:41: error: lvalue required as unary ‘&’ operand
   29 |                             , '#',11,&((vlSelfRef.test__DOT__test
      |                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~
   30 |                                         >> 3U))) ;
      |                                         ^~~~~~~
make: *** [/usr/local/share/verilator/include/verilated.mk:300: Vtest__ALL.o] Error 1
make: Leaving directory '/home/adrien/verilator_test/obj_dir'
%Error: 'make -C obj_dir -f Vtest.mk -j 1' exited with 2
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.053 for more assistance.
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/bin/verilator_bin --cc --timescale-override 1ps/1ps test.sv --timing --top-module test --binary
```

wsnyder, 2026-10-22:
File a new issue please, as was closed makes tracking new case easier.

adrienlemasle, 2026-10-22:
Done see #8860