verilator/verilator #3226: verilator lint feature that checks for port name suffixes

tushar-uwaterloo, 2021-12-05:
A good lint feature is to add suffixes in the port names. For instance, 

input port name must end with "_i or _in"
output port name must end with "_o or _out"
inout port name must end with "_io or _inout"

If I can get some assistance about where to start, I can start digging into it myself.

wsnyder, 2021-12-05:
You might want to alternatively look at verible, as it does pre-elaboration linting which has the name checks you mention.

I'm also receptive to adding name checks to Verilator lint.

Here's a list of other names that can/should be enforced (superset of Veriable).  Each would default to ".*" that is everything allowed.  Before implementing we should compare this list with Variable and other tools that have name checks.

* name_assert_regexp
* name_block_regexp
* name_class_regexp
* name_constraint_regexp
* name_cover_regexp
* name_enum_regexp
* name_enum_value_regexp
* name_function_regexp
* name_interface_regexp
* name_localparam_regexp
* name_macro_regexp
* name_member_regexp
* name_module_instance_regexp
* name_module_regexp
* name_package_regexp
* name_parameter_regexp
* name_port_in_regexp
* name_port_io_regexp
* name_port_out_regexp
* name_program_instance_regexp
* name_program_regexp
* name_signal_regexp
* name_struct_regexp
* name_task_regexp
* name_typedef_regexp
* name_union_regexp

To implement each of these
- [ ] Create a test_regress style testcase
- [ ] Add new VLT language parser rule to look for the format specification, e.g. in a .vlt file to setup these rules one would use

       name_typedef_regexp ".*_t$"

- [ ] Add new NAMING warning to V3Error
- [ ] Add dependency on libpcre so we can use that to handle matching regexps
- [ ] Add check to throw warning
- [ ] Add documentation

I'd suggest implement a single check, then we get the pull request looking good, then some copying will get us to the others relatively quickly.

DaveMcEwan, 2021-12-14:
Another pre-elaboration alternative (similar to Verible) is svlint: https://github.com/dalance/svlint
I'm adding on rules for naming conventions in my "free" time.