verilator/verilator #3228: Associate arrays: call to first() generates invalid C++ code

adrienlemasle, 2021-12-06:
The C++ code generated by the following is missing a semi-colon:
```
module test;

   logic [15:0] test [integer];

   initial begin
      integer index;
      test.first(index);
      $finish;
   end

endmodule
```

Output:
```
In file included from Vtest__ALL.cpp:6:
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp: In function ‘void Vtest___024root___initial__TOP__1(Vtest___024root*)’:
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp:14:78: error: expected ‘;’ before ‘VL_FINISH_MT’
   14 |     vlSelf->test__DOT__test.first(vlSelf->test__DOT__unnamedblk1__DOT__index)VL_FINISH_MT("test.sv", 8, "");

```

adrienlemasle, 2021-12-06:
Are all these associative arrays functions supposed to work. There is an issue with next(ref index) as well:

```
module test;

   logic [15:0] test [integer];

   initial begin
      integer index = 18;
      test[18] = 1;
      test[29] = 2;
      $display(test.next(index));
      $finish;
   end

endmodule
```

```
In file included from Vtest__ALL.cpp:6:
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp: In function ‘void Vtest___024root___initial__TOP__1(Vtest___024root*)’:
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp:16:56: error: cannot bind non-const lvalue reference of type ‘unsigned int&’ to an rvalue of type ‘unsigned int’
   16 |     VL_WRITEF("%11d\n",32,vlSelf->test__DOT__test.next(0x12U));
      |                                                        ^~~~~
In file included from /usr/local/share/verilator/include/verilated.h:876,
                 from Vtest.h:11,
                 from Vtest.cpp:4,
                 from Vtest__ALL.cpp:3:
/usr/local/share/verilator/include/verilated_types.h:572:21: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::next(T_Key&) const [with T_Key = unsigned int; T_Value = short unsigned int]’
  572 |     int next(T_Key& indexr) const {
      |              ~~~~~~~^~~~~~
make: *** [/usr/local/share/verilator/include/verilated.mk:253: Vtest__ALL.o] Error 1

In this particular case the call to next() in the C++ should pass a pointer to an integer, not an interger.
```

wsnyder, 2021-12-06:
These should work in more complicated cases; for this test I believe the bug is Verilator is incorrectly constant propagating in the value into the next call.

wsnyder, 2021-12-06:
>test.first(index);

And this bug is different, it doesn't properly remove a next() etc statement that has no effect.

Note both these bugs will probably have similar broken cases on all array, queue, enum, and associative methods.

wsnyder, 2021-12-06:
>test.first(index);
This is still a do-nothing statement.  As opposed to e.g.

     X = test.first(index)

adrienlemasle, 2021-12-06:
I believe index is passed by reference in this function so this is not really a do nothing
```
function int first (ref index);	Assigns to the given index variable the value of the first index; returns 0 for empty array
```

wsnyder, 2021-12-06:
Would you be willing to attempt a fix for this?  We need a bunch of new test cases for each method that check use as a statement, and a fix to V3Width to look for their use as a statement then either delete the statement (if has no effect) or properly call "makeStatement" which will eventually add the semicolon.

wsnyder, 2021-12-06:
Ah, you're right.  So the tests also need to check the ref'ed value is properly changed in these statement cases.

adrienlemasle, 2021-12-06:
Yes I'm happy to attempt a fix if you explain me how the C++ code works. Though if you have an idea about the other bug I've raised (the foreach bug), I'd rather attempt this one first as it is blocking for some of my code.

wsnyder, 2022-01-01:
Committed a fix for associative arrays, which is the place this is most likely to be needed.

The issue related to $display is separate, and related to fixing #3224.

closed 2022-01-01 by wsnyder