verilator/verilator #3229: Associative arrays: foreach silently not working

adrienlemasle, 2021-12-06:
```
module test;

   logic [15:0] test [integer];

   initial begin
      test[18] = 1;
      test[29] = 2;
      foreach(test[i]) begin
         $display("%0d", i);
      end
      $finish;
   end

endmodule
```

Expected output:
```
18
29
- test.sv:11: Verilog $finish
```

Actual output:
```
- test.sv:11: Verilog $finish
```

wsnyder, 2021-12-07:
Looking at the code support for foreach on associative, queues, and dynamic arrays was never added but should have been (or at least made an unsupported error).  Looking into it - unfortunately where it currently converts foreach makes it complicated, as does the desire to be able to unroll constant-bound foreach loops.

adrienlemasle, 2021-12-07:
Queues do seem to work though:

```
module test;

   logic [15:0] test [$];

   initial begin
      test.push_back(1);
      test.push_back(2);
      foreach(test[i]) begin
         $display("%0d %0d", i, test[i]);
      end
      $finish;
   end

endmodule
```

```
0 1
1 2
- test.sv:11: Verilog $finish
```

closed 2021-12-11 by wsnyder