verilator/verilator #3231: Listing package twice in command list results in broken link in node crash

bluewww, 2021-12-07:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```bash
$ cat pkg.sv
package pkg;
endpackage
$ cat top.sv
module top import pkg::*; #(
) (
);
endmodule
$ cat a.f
pkg.sv
$ cat b.f
pkg.sv
top.sv
```
When calling
`verilator --lint-only -f a.f -f b.f --debug --gdbbt`

I get
```
No stack.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Starting Verilator 4.216 2021-12-05 rev v4.216
- Verilator.cpp:559:  Option --verilate: Start Verilation
- V3ParseImp.cpp:273: parseFile: pkg
  Preprocessing pkg.sv
- V3PreShell.cpp:147:     Reading pkg.sv
- V3ParseImp.cpp:322: Lexing pkg.sv
- V3ParseImp.cpp:273: parseFile: pkg
  Preprocessing pkg.sv
- V3PreShell.cpp:147:     Reading pkg.sv
- V3ParseImp.cpp:322: Lexing pkg.sv
- V3ParseImp.cpp:273: parseFile: top
  Preprocessing top.sv
- V3PreShell.cpp:147:     Reading top.sv
- V3ParseImp.cpp:322: Lexing top.sv
%Warning-MODDUP: pkg.sv:1:9: Duplicate declaration of module: 'pkg'
    1 | package pkg;
      |         ^~~
                 pkg.sv:1:9: ... Location of original declaration
    1 | package pkg;
      |         ^~~
-node: PACKAGE 0xdfec40 <e9#> {e1aj} u4=0xdfed70  pkg  L0 [LIB] [NONE]
                 ... For warning description see https://verilator.org/warn/MODDUP?v=4.216
                 ... Use "/* verilator lint_off MODDUP */" and lint_on around source to disable this message.
dot -Tpdf -o ~/a.pdf obj_dir/Vpkg_001_linkcells.dot
- V3LinkLevel.cpp:45: modSortByLevel()
- V3Ast.cpp:1127:     Dumping obj_dir/Vpkg_002_cells.tree
%Error: Internal Error: top.sv:1:22: ../V3Broken.cpp:195: Broken link in node (or something without maybePointedTo): '!m_packagep || !m_packagep->brokeExists()' @ ../V3AstNodes.h:2663
-node: PACKAGEIMPORT 0xdf5830 <e12> {f1aw}  * -> PACKAGE 0xdfec40 <e15> {e1aj}  pkg  L3 [LIB] [NONE]
    1 | module top import pkg::*; #(
      |                      ^~
- V3Ast.cpp:1127:     Dumping obj_dir/Vpkg_990_final.tree
- V3Broken.cpp:343:   Broken called under broken, skipping recursion.
- V3StatsReport.cpp:220:statsReport: 
%Error: Internal Error: Aborting since under --debug

Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff7d2e323 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff7cde366 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff7cc77d3 in __GI_abort () at abort.c:79
#4  0x000000000062bc4b in V3Error::vlAbort () at ../V3Error.cpp:171
#5  0x000000000062c4d5 in V3Error::vlAbortOrExit () at ../V3Error.cpp:163
#6  0x000000000062e25f in V3Error::v3errorEnd (sstr=..., locationStr=...) at ../V3Error.cpp:287
#7  0x000000000065125c in FileLine::v3errorEnd (this=this@entry=0xdf5790, sstr=..., locationStr=...) at ../V3FileLine.cpp:359
#8  0x000000000042f18a in AstNode::v3errorEnd (this=this@entry=0xdf5830, str=...) at ../V3Ast.cpp:1208
#9  0x000000000042f277 in AstNode::v3errorEndFatal (this=this@entry=0xdf5830, str=...) at ../V3Ast.cpp:1151
#10 0x000000000046bd94 in BrokenCheckVisitor::processEnter (this=this@entry=0x7fffffffd190, nodep=nodep@entry=0xdf5830) at ../V3Error.h:312
#11 0x000000000046c3f3 in BrokenCheckVisitor::processAndIterate (nodep=0xdf5830, this=0x7fffffffd190) at ../V3Broken.cpp:220
#12 BrokenCheckVisitor::visit (this=0x7fffffffd190, nodep=0xdf5830) at ../V3Broken.cpp:326
#13 0x000000000040de7d in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:268
#14 0x000000000044952a in AstPackageImport::accept (this=<optimized out>, v=...) at ../V3AstNodes.h:2661
#15 0x000000000042cb9c in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:882
#16 0x000000000042cc2c in AstNode::iterateChildrenConst (this=this@entry=0xe03a40, v=...) at ../V3Ast.cpp:816
#17 0x000000000046c3fe in AstNVisitor::iterateChildrenConst (nodep=0xe03a40, this=0x7fffffffd190) at ../V3Ast.h:3001
#18 BrokenCheckVisitor::processAndIterate (nodep=0xe03a40, this=0x7fffffffd190) at ../V3Broken.cpp:221
#19 BrokenCheckVisitor::visit (this=0x7fffffffd190, nodep=0xe03a40) at ../V3Broken.cpp:326
#20 0x000000000040dc85 in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:240
#21 0x000000000040da33 in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:206
#22 0x00000000004494c4 in AstModule::accept (this=<optimized out>, v=...) at ../V3AstNodes.h:2581
#23 0x000000000042cb9c in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:882
#24 0x000000000042cc1b in AstNode::iterateChildrenConst (this=this@entry=0xde13a0, v=...) at ../V3Ast.cpp:815
#25 0x000000000046c3fe in AstNVisitor::iterateChildrenConst (nodep=0xde13a0, this=0x7fffffffd190) at ../V3Ast.h:3001
#26 BrokenCheckVisitor::processAndIterate (nodep=0xde13a0, this=0x7fffffffd190) at ../V3Broken.cpp:221
#27 BrokenCheckVisitor::visit (this=0x7fffffffd190, nodep=0xde13a0) at ../V3Broken.cpp:326
#28 0x000000000040db0b in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:218
#29 0x0000000000449e72 in AstNetlist::accept (this=<optimized out>, v=...) at ../V3AstNodes.h:9209
#30 0x0000000000468377 in AstNVisitor::iterate (nodep=0xde13a0, this=0x7fffffffd190) at ../V3Ast.h:2992
#31 BrokenCheckVisitor::BrokenCheckVisitor (nodep=0xde13a0, this=0x7fffffffd190) at ../V3Broken.cpp:331
#32 V3Broken::brokenAll (nodep=nodep@entry=0xde13a0) at ../V3Broken.cpp:347
#33 0x0000000000435963 in AstNode::dumpTreeFile (this=this@entry=0xde13a0, filename=..., append=append@entry=false, doDump=doDump@entry=true, doCheck=doCheck@entry=true) at ../V3Ast.cpp:1146
#34 0x000000000066a032 in V3Global::dumpCheckGlobalTree (stagename=..., newNumber=newNumber@entry=0, doDump=doDump@entry=true) at ../V3Global.cpp:92
#35 0x0000000000729cb1 in V3LinkLevel::modSortByLevel () at ../V3LinkLevel.cpp:84
#36 0x0000000000406ef7 in process () at ../Verilator.cpp:118
#37 0x000000000040951c in verilate (argString=...) at ../Verilator.cpp:602
#38 0x000000000040a06f in main (argc=<optimized out>, argv=0x7fffffffd858, env=<optimized out>) at ../Verilator.cpp:728
A debugging session is active.

        Inferior 1 [process 13846] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]
%Error: export VERILATOR_ROOT=
%Error: setarch --addr-no-randomize gdb /home/bluew/.local/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --lint-only -f a.f -f b.f --debug --gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed setarch --addr-no-randomize gdb /home/bluew/.local/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --lint-only -f a.f -f b.f --debug --gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'
```

What 'verilator --version' are you using?  Did you try it with the git master version?
`Verilator 4.216 2021-12-05 rev v4.216`

What OS and distribution are you using?
```
NAME="openSUSE Tumbleweed"
# VERSION="20211124"
```

Would you be willing to try to fix Verilator yourself with assistance?
Yes

wsnyder, 2021-12-07:
Not surprising bug in retrospect.  If you'd like to fix it that would be appreciated, first make a little test_regress module that shows the problem (it'll need an import), then I think in V3LinkCells.cpp right after where MODDUP is printed instead of deleting it, try setting "nodep->dead(true);"

bluewww, 2021-12-07:
thanks for the quick reply. I'm looking into how to do the test_regress thing

bluewww, 2021-12-08:
I added a test and implemented the fix you suggested.

Marking the node as dead instead of deleting in doesn't seem to do it. I still get a similar crash:

```
No stack.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Starting Verilator 4.216 2021-12-05 rev v4.216
- Verilator.cpp:559:  Option --verilate: Start Verilation
- V3ParseImp.cpp:273: parseFile: pkg
  Preprocessing pkg.sv
- V3PreShell.cpp:147:     Reading pkg.sv
- V3ParseImp.cpp:322: Lexing pkg.sv
- V3ParseImp.cpp:273: parseFile: pkg
  Preprocessing pkg.sv
- V3PreShell.cpp:147:     Reading pkg.sv
- V3ParseImp.cpp:322: Lexing pkg.sv
- V3ParseImp.cpp:273: parseFile: top
  Preprocessing top.sv
- V3PreShell.cpp:147:     Reading top.sv
- V3ParseImp.cpp:322: Lexing top.sv
%Warning-MODDUP: pkg.sv:1:9: Duplicate declaration of module: 'pkg'
    1 | package pkg;
      |         ^~~
                 pkg.sv:1:9: ... Location of original declaration
    1 | package pkg;
      |         ^~~
-node: PACKAGE 0xdefb90 <e9#> {c1aj} u4=0xdefcc0  pkg  L0 [LIB] [NONE]
                 ... For warning description see https://verilator.org/warn/MODDUP?v=4.216
                 ... Use "/* verilator lint_off MODDUP */" and lint_on around source to disable this message.
dot -Tpdf -o ~/a.pdf obj_dir/Vpkg_001_linkcells.dot
- V3LinkLevel.cpp:45: modSortByLevel()
- V3Ast.cpp:1127:     Dumping obj_dir/Vpkg_002_cells.tree
%Error: Internal Error: top.sv:1:22: ../V3Broken.cpp:195: Broken link in node (or something without maybePointedTo): '!m_packagep || !m_packagep->brokeExists()' @ ../V3AstNodes.h:2663
-node: PACKAGEIMPORT 0xdf1aa0 <e12> {d1aw}  * -> PACKAGE 0xdefb90 <e13> {c1aj}  pkg  L3 [LIB] [DEAD] [NONE]
    1 | module top import pkg::*; #(
      |                      ^~
- V3Ast.cpp:1127:     Dumping obj_dir/Vpkg_990_final.tree
- V3Broken.cpp:343:   Broken called under broken, skipping recursion.
- V3StatsReport.cpp:220:statsReport: 
%Error: Internal Error: Aborting since under --debug

Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff7d2e323 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff7cde366 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff7cc77d3 in __GI_abort () at abort.c:79
#4  0x000000000062bc4b in V3Error::vlAbort () at ../V3Error.cpp:171
#5  0x000000000062c4d5 in V3Error::vlAbortOrExit () at ../V3Error.cpp:163
#6  0x000000000062e25f in V3Error::v3errorEnd (sstr=..., locationStr=...) at ../V3Error.cpp:287
#7  0x000000000065125c in FileLine::v3errorEnd (this=this@entry=0xdf1a00, sstr=..., locationStr=...) at ../V3FileLine.cpp:359
#8  0x000000000042f18a in AstNode::v3errorEnd (this=this@entry=0xdf1aa0, str=...) at ../V3Ast.cpp:1208
#9  0x000000000042f277 in AstNode::v3errorEndFatal (this=this@entry=0xdf1aa0, str=...) at ../V3Ast.cpp:1151
#10 0x000000000046bd94 in BrokenCheckVisitor::processEnter (this=this@entry=0x7fffffffd170, nodep=nodep@entry=0xdf1aa0) at ../V3Error.h:312
#11 0x000000000046c3f3 in BrokenCheckVisitor::processAndIterate (nodep=0xdf1aa0, this=0x7fffffffd170) at ../V3Broken.cpp:220
#12 BrokenCheckVisitor::visit (this=0x7fffffffd170, nodep=0xdf1aa0) at ../V3Broken.cpp:326
#13 0x000000000040de7d in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:268
#14 0x000000000044952a in AstPackageImport::accept (this=<optimized out>, v=...) at ../V3AstNodes.h:2661
#15 0x000000000042cb9c in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:882
#16 0x000000000042cc2c in AstNode::iterateChildrenConst (this=this@entry=0xdf1740, v=...) at ../V3Ast.cpp:816
#17 0x000000000046c3fe in AstNVisitor::iterateChildrenConst (nodep=0xdf1740, this=0x7fffffffd170) at ../V3Ast.h:3001
#18 BrokenCheckVisitor::processAndIterate (nodep=0xdf1740, this=0x7fffffffd170) at ../V3Broken.cpp:221
#19 BrokenCheckVisitor::visit (this=0x7fffffffd170, nodep=0xdf1740) at ../V3Broken.cpp:326
#20 0x000000000040dc85 in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:240
#21 0x000000000040da33 in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:206
#22 0x00000000004494c4 in AstModule::accept (this=<optimized out>, v=...) at ../V3AstNodes.h:2581
#23 0x000000000042cb9c in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:882
[... 1 lines cut ...]
#25 0x000000000046c3fe in AstNVisitor::iterateChildrenConst (nodep=0xde13a0, this=0x7fffffffd170) at ../V3Ast.h:3001
#26 BrokenCheckVisitor::processAndIterate (nodep=0xde13a0, this=0x7fffffffd170) at ../V3Broken.cpp:221
#27 BrokenCheckVisitor::visit (this=0x7fffffffd170, nodep=0xde13a0) at ../V3Broken.cpp:326
#28 0x000000000040db0b in AstNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor.h:218
#29 0x0000000000449e72 in AstNetlist::accept (this=<optimized out>, v=...) at ../V3AstNodes.h:9209
#30 0x0000000000468377 in AstNVisitor::iterate (nodep=0xde13a0, this=0x7fffffffd170) at ../V3Ast.h:2992
[... cut ...]
%Error: setarch --addr-no-randomize gdb /home/bluew/projects/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --lint-only pkg.sv pkg.sv top.sv --debug -gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed setarch --addr-no-randomize gdb /home/bluew/projects/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --lint-only pkg.sv pkg.sv top.sv --debug -gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'
```

Do you have other suggestions where I should look at? Maybe the duplicate module checking needs to special case for packages? I don't have yet a good view where things happend in the codebase.

wsnyder, 2021-12-10:
When you run with --debug and look at the tree file, is the PACKAGEIMPORT that later gives the error under the dead package, or is a non-dead package picking up the import?

wsnyder, 2023-08-30:
Fixed a while ago - example passes after MODDUP suppressed - see above comments for commit info.

closed 2023-08-30 by wsnyder