verilator/verilator #3232: Wrong behaviour of VL_STREAML_FAST_QQI with Intel Compiler when lbits=64

adrienlemasle, 2021-12-08:
The Intel compiler seems to optimise differently from GCC on shifting up a QData by 64 bits (which creates an overflow). For example, for Intel compiler (~0ULL)<<64 and ~((~0ULL)<<64) are both 0.

This creates an issue in VL_STREAML_FAST_QQI which always ends up returning 0 when lbits=64.

A fix for this is to change this line in VL_STREAML_FAST_QQI function definition in verilated_funcs.h:
```
        const QData msbMask = lbitsFloor == VL_MASK_Q(lbitsRem) << lbitsFloor;
```
to:
```
        const QData msbMask = lbitsFloor == 64 ? 0ULL : VL_MASK_Q(lbitsRem) << lbitsFloor;
```

adrienlemasle, 2021-12-08:
I'm not sure whether this is an actual Intel compiler bug or the C++ spec allows to do whatever in case of overflows. The behaviour above is with O3. With O0 Intel compiler also does something different from what we would expect (I haven't debugged O0 yet though).

wsnyder, 2021-12-08:
Either behavior is legal in C/C++, there's other similar workarounds already to account for this.

Do you have a simple test case that shows the problem, or does it show up as part of the existing tests?

Can you make this into a pull request?

adrienlemasle, 2021-12-08:
I haven't run the existing tests with Intel Compiler actually. I should try. But I can easily generate a test case for this if it doesn't already exist.

Sure I will make a pull request.

closed 2021-12-09 by wsnyder