verilator/verilator #3233: UNOPTTHREAD when compile a large design

EDAHUB, 2021-12-08:
We are running multi thread --thread 8 and met the following issue:

Parallelism estimate for based on mtask costs:
Critical path cost: 1300000
Total graph cost = 132987452
MTask vertex count = 12084764
Edge count = 27649816
Parallelism factor = 98
%Warning-UNOPTTHREADS: Thread scheduler is unable to provide requested parallelism; suggest asking for fewer threads.
V3partition.cpp: 1160: mtaskCount = 6512 maxMTask=400

With that we increased with --threads-max-mtasks to 8000 and recompiling. But More MTasks will result in more communication and synchronization overhead at simulation runtime. This design is large, clock tree is fairly simple. Any suggestion on how to reduce the mtaskCount number?

Bye the way, copied from V3Partition.cpp. If mtaskCount can not be less than maxMTasks, will this while(true) be a dead loop? We are seeing long compile time when maxMtask is set to 400 in the previous compile.

1104          while(true) {
   ......
1151                     if (mtaskCount > maxMTasks) {
1152                         uint32_t oldLimit = m_scoreLimit;
1153                         m_scoreLimit = (m_scoreLimit * 120) / 100;
1154                         v3Global.rootp()->fileline()->v3warn(
1155                             UNOPTTHREADS, "Thread scheduler is unable to provide requested "
1156                                           "parallelism; suggest asking for fewer threads.");
1157                         UINFO(1, "Critical path limit was=" << oldLimit << " now=" << m_scoreLimit
1158                                                             << endl);
1159                         continue;
1160                     }

wsnyder, 2021-12-09:
I believe because the current warning increases the limit, and all other continue's are just to rescore the loop should terminate. But I might be reading it wrong, did your compile ever finish after printing the warning?

I suspect there's a giant mtask causing the problem as opposed to too many mtasks.  It would be interesting if you just disabled the warning and see how the gantt chart ends up - you have a fair number of mtasks as it is, so it might turn out ok.  (In which case we should consider better tuning the warning message.)

EDAHUB, 2021-12-09:
With the warning message, we did not get successful compile done.

Will try to create a test case that can be shared and update status again.

wsnyder, 2022-05-12:
I'm presuming this got figured out, otherwise please provide a test case thanks.

closed 2022-05-12 by wsnyder