verilator/verilator #3236: Call to data() function on non class type (WData)

adrienlemasle, 2021-12-09:
Another one in my bug finding spree :)

```
module test(
            input clk,
            input reset,
            output logic [95:0] a
            );

   string         a_s;

   initial begin
      $sformat(a_s, "%h", a);
      $display(a_s);
   end

endmodule
```
Leads to wrong C++:
```
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp:15:28: error: request for member ‘data’ in ‘vlSelf->Vtest___024root::a’, which is of non-class type ‘WData [3]’ {aka ‘unsigned int [3]’}
   15 |                  vlSelf->a.data());

```

adrienlemasle, 2021-12-09:
I think the bug is in V3EmitFunc.h in this code:
```
    void emitDatap(AstNode* nodep) {
        // When passing to a function with va_args the compiler doesn't
        // know need a pointer so when wide, need to look inside VlWide
        if (nodep->isWide()) puts(".data()");
    }
```
VL_OUTW is wide but is not a VlWide. Not sure how to fix this though.

wsnyder, 2021-12-09:
BTW it's a little surprising this wasn't noticed until now...

adrienlemasle, 2021-12-09:
This is not a super common case I guess. This will only happen calling a format/print type function with IOs of 64-bit or more as argument.

Yes when I tried declaring wide primary IOs as VlWide instead of WData it fixed the issue. But I was thinking the WData was used there for a good reason, backwards compatibility being one.

wsnyder, 2021-12-10:
I think it'll be less of a future bug farm to accept changing the primary I/O types to use VlWide.  When I do that all the existing tests pass, so shouldn't be too bad on users.

closed 2021-12-10 by wsnyder