verilator/verilator #3240: SystemC AMS

mzau, 2021-12-10:
Does verilator work with SystemC AMS, for example the proof-of-concept implementation from accellera.org (Coseda GmbH) ?

wsnyder, 2021-12-10:
Short answer, no.
Longer answer, mostly no, see https://veripool.org/guide/latest/languages.html#verilog-ams-support

closed 2021-12-10 by wsnyder

barnasc, 2021-12-11:
@wsnyder: SystemC AMS is something different than Verilog-AMS.

@mzau : Verilator can generate code embedded in a SystemC module with SystemC input and output ports (see https://verilator.org/guide/latest/connecting.html#connecting-to-systemc). 
As a next step, you can simply connect your SystemC-AMS model, which is basically also C++ based, to the SystemC model generated by Verilator. To simulate the whole thing, you can use the Accellera reference implementation for SystemC and the SystemC-AMS Proof-of-concept simulator offered by COSEDA Technologies.

wsnyder, 2021-12-11:
Thanks for the update - wasn't aware of SC AMS.

mzau, 2021-12-13:
Sorry, that my question wasn't exact enough  - I didn't mean to verilate Verilog-AMS code, but pure synthesizeable SystemVerilog without any AMS features. My understanding maybe not deep enough - so the questions, which appearing to me: Is the verilated SystemC code a "true SystemC citizen", like any other non AMS related synthesizeable SystemC code? Or is there some interaction with the time model, which probably the systemc-ams lib from COSEDA can't get aware of?  My current problem is, that I have a SystemC dut, which works with a SystemC-AMS model - but when I convert it to SystemVerilog and  use the back verilated SystemC code, the AMS model solver doesn't work. Essentially I need some hints to work out if the problem is a verilator or systemc-ams issue.

wsnyder, 2021-12-13:
The Verilated model should be a "pure" SystemC citizen.  The only slight hackage is that for SystemC tracing Verilator has some special hooks to get the trace calls into its native form.

barnasc, 2021-12-13:
The SystemC-AMS solvers/schedulers work on top of the SystemC discrete event engine, so assuming you use the Accellera SystemC reference implementation in both cases, I do not expect it is a simulation kernel issue, but a modeling issue. Note that the synchronization between SystemC discrete event signals and SystemC-AMS Timed Data Flow signals are "loosely coupled" so if you are experiencing issues related to cycle-accurate timing, you first need to revisit the timing deltas/accuracy between the original SystemC model and the Verilated SystemC model, and secondly check if any timing difference might affect signals passed from/to the SystemC AMS model.

Einwich, 2021-12-13:
I can confirm, that SystemC AMS behaves from the SystemC point of view like one or more SystemC processes - thus like any other SystemC model. SystemC AMS is build on top of the SystemC standard and did not uses backdoors. Thus there is from the kernels point of view no issue. We are using SystemC AMS in conjunction with verilated Verilog models and saw up to now no problem.

mzau, 2021-12-13:
@Einwich Thank you for clarifying that.