verilator/verilator #3241: Parameter not properly overwritten by -G option if an imported package contains a parameter with the same name

adrienlemasle, 2021-12-10:
Verilog code:
```
package pack_a;
   parameter PARAM_A = 0;
endpackage : pack_a

import pack_a::*;

module test(
            input clk,
            input reset
            );

   parameter PARAM_A = 0;

   initial begin
      $display(PARAM_A);
      $finish;
   end

endmodule
```

Command line:
```
verilator --cc --timescale-override 1ps/1ps test.sv --top-module test --exe --build main.cpp -GPARAM_A=1
```

Result:
```
          0
- test.sv:16: Verilog $finish
```

Expected result:
```
          1
- test.sv:16: Verilog $finish
```

wsnyder, 2021-12-10:
PARAM_A isn't a top level parameter, it's a parameter under pack_a.  An import just makes it easier to access from the top it doesn't make it "belong" to the top, it's like a shorthand for writing "pack_a::PARAM_A" in your code.  Hence -G won't work on it.

wsnyder, 2021-12-10:
But there's no import in the module, missed that at first.

wsnyder, 2021-12-10:
Perhaps you'd be willing to look into the code and see what is happening?

adrienlemasle, 2021-12-10:
You can see I am redefining the parameter inside the module itself. The expected behaviour I've shown is how Questa, Riviera and VCS behave.

adrienlemasle, 2021-12-10:
> 

> Perhaps you'd be willing to look into the code and see what is happening?

Yes I will have a look at the code to see what is happening.

wsnyder, 2021-12-11:
Thanks for making a pull request!

closed 2021-12-11 by wsnyder