verilator/verilator #3242: UNUSED is mistakenly reported on signal connected to inout port of two submodules

quark17, 2021-12-10:
The attached example ([inout_connect_inlined.tar.gz](https://github.com/verilator/verilator/files/7695335/inout_connect_inlined.tar.gz)) has a Sender and Receiver submodule each with an `inout` port, through which the sender writes a value and the receiver reads it.  These are instantiated in a parent module that wires them together:
```
wire [31 : 0] recv$IN;
Receiver recv(.CLK(CLK), .IN(recv$IN));
Sender #(.outvalue(32'd3737)) send(.OUT(recv$IN));
```
Without `-Wall`, Verilator compiles and runs the example (transmitting the value 3737 from sender to receiver):
```
      3737
      3737
```
With warnings turned on, Verilator reports that the connecting wire is unused:
```
%Warning-UNUSED: mod.v:11:17: Signal is not used: 'recv$IN'
                            : ... In instance mod
   11 |   wire [31 : 0] recv$IN;
      |                 ^~~~~~~
                 ... For warning description see https://verilator.org/warn/UNUSED?v=4.215
                 ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message.
```
This happens with a recent HEAD and an earlier release:
```
Verilator 4.215 devel rev v4.214-50-ga14394db
Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)
```

wsnyder, 2021-12-10:
UNUSED is performed module-by-module so it won't presently do well detecting tristate problems.  My inclination is to suppress UNUSED on any connection to a submodule with an inout.

wsnyder, 2022-01-01:
Fixed by adding hack to consider all inout pins driven.

closed 2022-01-01 by wsnyder