verilator/verilator #3243: BLKANDNBLK warning reported for a non-assignment use

quark17, 2021-12-10:
The attached example ([error-BLKANDNBLK.tar.gz](https://github.com/verilator/verilator/files/7695690/error-BLKANDNBLK.tar.gz)) gives this error:
```
%Error-BLKANDNBLK: mod.v:6:16: Unsupported: Blocked and non-blocking assignments to same variable: 'mod.fr'
    6 |   reg [31 : 0] fr;
      |                ^~
                   mod.v:62:21: ... Location of blocking assignment
   62 |    b__h333 = $fgetc(fr);
      |                     ^~
                   mod.v:39:16: ... Location of nonblocking assignment
   39 |     if (fr$EN) fr <= fr$D_IN;
      |                ^~
                   ... For error description see https://verilator.org/warn/BLKANDNBLK?v=4.215
```
claiming that `fr` is assigned in two different ways, but the location that it points to for the nonblocking assignment is not an assignment to `fr` (though `fr` is used in the RHS).  Am I missing some reason that this is being treated as an assignment?

This happens with a recent HEAD and an earlier release:
```
Verilator 4.215 devel rev v4.214-50-ga14394db
Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)
```

wsnyder, 2021-12-10:
$fgetc modifies fr, so is effectively an assignment.

In general using blocking assignments with a file handle is probably asking for trouble as file handles shouldn't have multiple simultaneous values (the previous and new values) which come from using them in a blocking manor.  Though there's no specific warning for that there perhaps should be.

closed 2021-12-10 by wsnyder

quark17, 2021-12-10:
Why do you say that `$fgetc` modifies `fr`?  Is there something in the spec that says that `fr` may be modified?  (And if I put `$display` of `fr` before and after the call, I see in Icarus and in Verilator that the value does not change.)

wsnyder, 2021-12-10:
Badly phrased.  The thing being pointed to is being modified - the file itself.  This is considered a change to the object as far as Verilator is considered, because this establishes what statements have dependencies on the variable.   A full IEEE compliant simulator doesn't rip apart procedures like Verilator does, so this is outside IEEE.

rsnikhil, 2023-07-16:
I don't understand the explanation above.

'fr' itself is not being modified by fgetc(), and that's all that should matter, even if it happens to be a handle to a file which is indeed being modified.

Attached is another small Verilog program, mkTop.v, using a register containing a file handle, and again the BLKANDNBLK error is puzzling.  [Thanks to @quark17 for boiling this down to a small example.]

In mkTop.v there are four fragments labeled v1, v2, v3 and v4, with only one fragment un-commented out.
For v1 and v3, verilator does not raise the BLKANDNBLK error.
For v2 and v3, verilator does raise the BLKANDNBLK error.

I'm compiling this with 'verilator --binary mkTop.v'
Version: Verilator 5.005 devel rev v5.004-63-g46cc7b5d0

[example.tar.gz](https://github.com/verilator/verilator/files/12063098/example.tar.gz)

quark17, 2023-07-16:
It's worth noting two things about the example just posted.

One, the error message is bad for v2 (the example as is).  If you compile with `verilator --binary mkTop.v`, you get this error:
```
%Error-BLKANDNBLK: mkTop.v:7:16: Unsupported: Blocked and non-blocking assignments to same variable: 'mkTop.logf'
    7 |   reg [31 : 0] logf;
      |                ^~~~
                   mkTop.v:47:15: ... Location of blocking assignment
   47 |      v__h3986 = v__h3992 / 32'd10;
      |               ^
                   mkTop.v:28:15: ... Location of nonblocking assignment
   28 |  if (logf$EN) logf <=  logf$D_IN;
      |               ^~~~
                   ... For error description see https://verilator.org/warn/BLKANDNBLK?v=5.006
```
where it points to line 47 that doesn't even include a mention of `logf`!

Two, the "blocking assignment" that Verilator reports in the error for v4 is `$fflush(logf)`.  The example v3 also has the `$fflush(logf)`, but no error is reported!  The difference is that there is also a reference to `logf` in the if-statement guarding the task.  That is, this code triggers an error:
```
  always@(negedge CLK)
  begin
     if (logf != 32'd0) $fflush(logf);
  end
```
but this code does not:
```
  always@(negedge CLK)
  begin
     $fflush(logf);
  end
```
And an if-statement referring to some other signal doesn't cause an error either, because this is fine:
```
  always@(negedge CLK)
  begin
     if (logf$EN == 1'd1) $fflush(logf);
  end
```
So it's unclear what rules Verilator is using to decide what is BLKANDNBLK or not.

Similarly, the difference between v1 (which doesn't error) and v2 (which errors), is just the addition of `$fflush`:
```
  always@(negedge CLK)
  begin
     v__h3992 = $stime;
     v__h3986 = v__h3992 / 32'd10;
     $fdisplay(logf, "%0d", v__h3986);
  end
```
vs
```
  always@(negedge CLK)
  begin
     v__h3992 = $stime;
     v__h3986 = v__h3992 / 32'd10;
     $fdisplay(logf, "%0d", v__h3986);
     $fflush(logf);
  end
```
But the first one still includes a `$fdisplay` to `logf` -- and if `$fgetc` is considered a "blocking assignment" on its filehandle, then `$fdisplay` should be considered that as well.  But Verilator isn't giving an error for v1, it is only considering the example to be BLKANDNBLK if `$fflush` is added (v2).  And, further, `$fflush` by itself (v3) is not enough to trigger the error, without some other reference to `flog` nearby -- either as the argument to `$fdisplay` (v2) or in the if-statement guard (v4).  All of which doesn't make sense.  And, if nothing else, you'll want to improve the error message for v2, shown above -- but I wonder if that is just evidence that Verilator's rules for deducing BLKANDNBLK are somehow wrong?