verilator/verilator #3244: %Error-UNSUPPORTED: Unsupported: enum next/prev method on enum with > 10 bits

agrobman, 2021-12-11:
Are there any plans to support enum values up to 32 bits at least?

wsnyder, 2021-12-11:
Not currently as it uses a lookup table.  Anything wider would use too much memory or be very slow.

Can you provide an example where the wide enums with next()/prev() are really the only good way to code something?  (Versus someone being lazy and using a wider type than needed.)

agrobman, 2021-12-11:
we have following type defined ( RISCV CSR definition): 
```
typedef enum logic [17:0] {
			   misa =18'h301,
			   mvendorid =18'hf11,
			   marchid =18'hf12,
			   mimpid =18'hf13,
			   mhartid =18'hf14,
			   sstatus =18'h100,
...
} csr_t;
```
and  TB uses it to initialize string array for disassembler:

    while(csr != csr.last)begin
        csr_name[csr] = csr.name;
        csr = csr.next;
    end

wsnyder, 2021-12-11:
That's certainly reasonable.  The fix would be it needs to implement a map to use over the current size limit.  Will add it to the list.

agrobman, 2021-12-11:
our issue is that these definitions are used in many places of the TB and  CSR content is changing from project to project pretty often

closed 2021-12-11 by wsnyder