verilator/verilator #3245: Resolve width mismatch of biops

yTakatsukasa, 2021-12-11:
As in https://github.com/verilator/verilator/pull/3221#issuecomment-982643245, inputs of BiOps don't necessarily have same bit width.

I added a simple visitor to compare inputs of BiOps, mismatch appears after constifyAllLint().
https://github.com/verilator/verilator/blob/748fa4cb8039971a2d5dc472923141846b8a3e87/src/Verilator.cpp#L211

I will narrow down which optimization introduces the mismatch.

yTakatsukasa, 2021-12-16:
[diary] I don't have much bandwidth lately, but gradually fixing mismatches.

Regression tests don't show bit width mismatch until PRE-FLAT OPTIMIZATION on my local branch.
https://github.com/verilator/verilator/blob/5e3016f34af6ed81bca4473d45a01ad5f118e865/src/Verilator.cpp#L248

Will need to decide the rule for width of constants.
1. Constants have minimum width to represent the value as described in V3Width. It means width mismatch will be allowed if one of the arguments is a constant.
  https://github.com/verilator/verilator/blob/5e3016f34af6ed81bca4473d45a01ad5f118e865/src/V3Width.cpp#L24
2. Constants have the same width as the other argument of a binary op.

gezalore, 2021-12-17:
> Will need to decide the rule for width of constants.

We should go with same with as other argument. Sometimes you can prove things about the variable that allow simplification, which is simpler to implement if you can just replace the node with one of it's operands.