verilator/verilator #3246: SystemC trace example does not produce valid vcd trace

derchr, 2021-12-11:
How may we help - what is your question?

Hello,
I don't know if I am doing something wrong but the `cmake_tracing_sc` does not produce a valid vcd dump. I run the example with the `+trace` flag and this is the vcd file that is created:
```
$version Generated by VerilatedVcd $end
$date Sat Dec 11 12:36:00 2021 $end
$timescale 1ps $end

 $scope module top $end
  $scope module top $end
   $var wire  1 # clk $end
   $var wire  1 $ fastclk $end
   $var wire 40 - in_quad [39:0] $end
   $var wire  2 , in_small [1:0] $end
   $var wire 70 / in_wide [69:0] $end
   $var wire 40 ' out_quad [39:0] $end
   $var wire  2 & out_small [1:0] $end
   $var wire 70 ) out_wide [69:0] $end
   $var wire  1 % reset_l $end
   $scope module sub $end
    $var wire  1 # clk $end
    $var wire 32 3 count_c [31:0] $end
    $var wire 32 2 count_f [31:0] $end
    $var wire  1 $ fastclk $end
    $var wire  1 % reset_l $end
   $upscope $end
  $upscope $end
 $upscope $end
$enddefinitions $end
```
As you can see, the signals get declared but the actual tracing is missing.
When I run the `cmake_tracing_c` example, everything works as expected and I can open the trace in GTKWave.
This is my first time using Verilator so it could be that I am missing something.

Thank you for your help!

wsnyder, 2021-12-11:
Does make_tracing_c (versus the cmake version) produce correct results or is this also broken?

What version of Verilator, SystemC and what OS distro are you using?

derchr, 2021-12-11:
> Does make_tracing_c (versus the cmake version) produce correct results or is this also broken?

With the `make_hello_c` example, it works. But with the `make_hello_sc` it does not work.

> What version of Verilator, SystemC and what OS distro are you using?

I've built Verilator from current master branch. SystemC is the newest version (2.3.3), built with CMake.
My OS is OpenSUSE Tumbleweed.

wsnyder, 2021-12-11:
Hmm, that's unexpected that make worked.  Perhaps look at the difference in the compiler command lines between using make and cmake?

derchr, 2021-12-11:
> Hmm, that's unexpected that make worked.

Hm, from my findings the Make version did not work either, I would say Make vs CMake is not the problem.
To sum it up: Both `cmake_tracing_sc` and `make_tracing_sc` did not work. But both `cmake_tracing_c` and `make_tracing_c` do work.

wsnyder, 2021-12-11:
Are you sure time is advancing - did the program finish?  Add print statements around the calls to tfp->flush and tfp->close in sc_main.cpp.

derchr, 2021-12-11:
I found the mistake.
This is from the FAQ:

> When using SystemC 2.3, the SystemC library must have been built with the experimental simulation phase callback based tracing disabled. This is disabled by default when building SystemC with its configure based build system, but when building SystemC with CMake, you must pass -DENABLE_PHASE_CALLBACKS_TRACING=OFF to disable this feature.

With recompiling SystemC with the new flag, it works.
Thanks for your help anyways!

closed 2021-12-11 by derchr