verilator/verilator #3248: Standalone reference to 'this' is marked as a symbol-resolution error

mballance, 2021-12-12:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

I've modified t_class_uses_this.v to show the issue. Specifically, I've added a method that sets the 'addr' field by assigning 'this' to a temp variable, and assigning 'addr' via this temp variable (see set_indirect below):

```
class Cls;
   bit [3:0] addr;
   function void set(bit [3:0] addr);
   begin : body
     this.addr = addr;
   end : body
   endfunction

   function void set_indirect(bit [3:0] addr);
   begin : body
     Cls c = this;
     c.addr = addr;
   end : body
   endfunction

   extern function void setext(bit [3:0] addr);
endclass
```

When I do this, Verilator reports:
```
%Error: t/t_class_uses_this.v:16:14: Can't find definition of variable: 'this'
   16 |      Cls c = this;
      |              ^~~~
%Error: Exiting due to 1 error(s)
%Warning: vlt/t_class_uses_this: Exec of perl failed: %Error: t/t_class_uses_this.v:16:14: Can't find definition of variable: 'this'
```

What 'verilator --version' are you using?  Did you try it with the git master version?
I'm using the latest Git source.

What OS and distribution are you using?
CentOS 8

Would you be willing to try to fix Verilator yourself with assistance?
Yes, would be happy to. I've been able to recognize the 'this' reference in "visit(AstParseRef* nodep)". I tried to adapt the this-reference code from "visit(AstDot* nodep)", but this results in:
```
%Error: Internal Error: t/t_class_uses_this.v:16:14: ../V3Width.cpp:4812: Visit function missing? Widthed expectation for this node: PARSEREF 0x2340f10 <e343#> {d16ao}  this [$root]
```

I suspect that I need to make a nodep->replaceWith() call, but I'm uncertain what to pass to provide the representation of the class instance.

wsnyder, 2021-12-12:
Note also #2594.

The internal error is because the ParseRef is getting through V3LinkDor into V3Width which it doesn't understand.  I think we want to change e.g. "this.addr" to a VarXRef that points to "addr" under "this.". 

For a standalone "this" I can see a few routes, a AstVarRef with name "this" - but we'd also need an AstVar 
I suspect we need a new AstThis node type, as it acts quite different from a normal variable.  That new AstThis will need a lot of plumbing throughout the various visitor stages - every place there's an AstVar.

wsnyder, 2021-12-12:
[edited]

Note also #2594.

The internal error is because the ParseRef is getting through V3LinkDor into V3Width which it doesn't understand. I think we want to change e.g. "this.addr" to a VarXRef that points to "addr" under "this.".

For a standalone "this" I think we'd want a AstVarRef with name "this" - and we'd also need an implicit AstVar created in each method that uses "this" so the AstVarRef has something to point to.  We'd also need AstVar::isThis() method as there will be some special checks and handling needed, for example it should never be allowed to be assigned by users.

wsnyder, 2021-12-12:
Actually I think this.addr is really like any other member reference, so it would be a MemberSel off from the new VarRef with name "this".

mballance, 2021-12-12:
Appreciate the feedback above, Wilson. Looks like the preferred path is to add a built-in 'this' variable to methods, along with some other checks. Any chance there is an existing case that is implemented as this sort of implicit variable? Hoping to find something to use as a pattern to copy...

wsnyder, 2021-12-12:
I think "randomize" is somewhat similar.

As to an example of method calls run e.g. "test_regress/t/t_class_method.pl --debug --dump-treei 9" and look at the tree file test_regress/obj_dir/.....007_linkdot.tree

So what we'd expect in the end is something like this:

```
    declaring "this" when used in a function:
    1: PACKAGE 0x555556bea120 <e208> {a0aa}  __024unit  L3 [LIB] [1ps]
    1:2: CLASS 0x555556be8180 <e51> {d9ab}  Cls
    1:2:2: FUNC 0x555556c10000 <e20> {d11ar} @dt=0@  get_methoda
    1:2:2:1: VAR 0x <e56> {d18al} @dt=0@  this

    then using this.member becomes roughly:
    1:2:2:2:2: MEMBERSEL 0x <e254#> {d21aj} @dt=0@  imembera -> %Error:UNLINKED
    1:2:2:2:2:1: VARREF 0x <e249#> {d21ah} @dt=0@  this [RV] <- VAR 0x <e56> {d18al} @dt=0@  this

    then using this.method() becomes roughly:
    1:2:2:2:3: METHODCALL 0x <e278#> {d23aj} @dt=0@  set_methoda ->  -> UNLINKED
    1:2:2:2:3:1: VARREF 0x <e249#> {d21ah} @dt=0@  this [RV] <- VAR 0x <e56> {d18al} @dt=0@  this
```

benjaminmordaunt, 2022-09-13:
I'm trying to have a stab at this and decipher the API of Verilator. (It would be awesome if there was a little more docs for internal, but I know docs for FOSS projects is hard work!)

I'm trying to understand the parallels between `V3Randomize` and what could be created in a `V3This`. One issue is that the `randomize()` method gets added to the class node, but I need a VarRef pointing to the current _instance_, instead of the class node. I'm struggling to understand how _instances_ present themselves in the code base. Node pointers are not instances, right? They're nodes in the AST tree (i.e. map to the language constructs...)

Any help would be appreciated @wsnyder .

wsnyder, 2022-09-13:
We don't need a new visitor for "this".  I suspect we don't even need to link it as a Var and can just pass through.

- [ ] Check that appropriate tests exist and pass on other simulators.  Include "failing" tests also e.g. if "this" is used in static function or outside a class.
- [ ] Probably need new AstThis in V3AstNodes.h
- [ ] Fix verilog.y to make sure parses ok; use --debug to look at tree files to see that looks like what expects
- [ ] Fix V3Width.cpp to handle the new AstThis
- [ ] Fix V3EmitC to output AstThis as "this"
- [ ] Not sure what else is needed, but as debug sure will find other things

benjaminmordaunt, 2022-09-13:
Hi, I've made good progress with this. The following sample (the original sample augmented with a should-fail test):
```
class Cls;
   bit [3:0] addr;
   function void set(bit [3:0] addr);
   begin : body
     this.addr = addr;
   end : body
   endfunction

   function void set_indirect(bit [3:0] addr);
   begin : body
     Cls c = this;
     c.addr = addr;
   end : body
   endfunction

   extern function void setext(bit [3:0] addr);
endclass

module test;

Cls c = this;

endmodule
```
... gives the following output...
```
%Error: test1.sv:16:25: definition not found for extern 'setext'
   16 |    extern function void setext(bit [3:0] addr);
      |                         ^~~~~~
%Error: test1.sv:21:9: 'this' used outside class (IEEE 1800-2017 8.15)
   21 | Cls c = this;
      |         ^~~~

```

I believe this is now the expected response? Verilator also continues to pass all other tests. I will write a few more tests and contribute.

wsnyder, 2022-09-13:
>I believe this is now the expected response?

Looks reasonable to me.  Good to check it with another simulator too if you have access to one.

mballance, 2022-09-21:
One of the tricks with implementing 'this' comes about on the code-gen side. 'this' is a reference to a class-type object in the same way that a variable is, so assigning 'this' to a variable must increment the object ref count. 
I had experimented with the following code for doing this. There are two parts:
- Additions to code generated to implement a SV class, allowing it to hold a self-reference
- A vl-specific version of 'make_shared' to properly set the class' self-reference

```
class C {
public:
        C(const std::string &name) : m_name(name) { }
        ~C() {
                fprintf(stdout, "~C %s\n", m_name.c_str());
        }

        void this_p(std::shared_ptr<C> tp) {
                m_tp = tp;
        }

        std::shared_ptr<C> this_p() const {
                return std::shared_ptr<C>(m_tp);
        }

private:
        std::string             m_name;
        std::weak_ptr<C>        m_tp;
};
```

The C++ code generated to implement a SV class needs to have  std::weak_ptr<> to hold a self-reference (above).

```
template <class T, class... P> std::shared_ptr<T> vl_make_shared(P... args) {
        std::shared_ptr<T> ret = std::make_shared<T>(args...);
        ret->this_p(ret);
        return ret;
}
```

The Verilator-specific version of make_shared simply sets the weak reference inside the class instance before returning it.

Hope this is helpful in implementing the back-end portion!

kbieganski, 2022-09-29:
Note that Verilator 5 no longer uses `shared_ptr` (https://github.com/verilator/verilator/pull/3595). Now there's a custom intrusive pointer (`VlClassRef`) which can be created from `this`, as it stores the ref count within the class instance. So the codegen part is pretty much handled there.

kozdra, 2022-10-13:
I started some work on this, you can see the changes on the branch for now: https://github.com/antmicro/verilator-1/tree/standalone-this

PR #3675 (2022-10-14): Support standalone 'this' in classes
I've seen some progress on the related issues, but it turned out fairly straightforward to implement, even with the new strict class checking.

Closes #2594
Closes #3248
closed 2022-10-14 by wsnyder